espressif/arduino-esp32 #4160: timerReadMicros returns older value

ummarbhutta, 2020-07-12:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I am working on an application where I have GPIO interrupt defined, and on each interrupt I want to record timer value for which I am calling `timerReadMicros` but the problem I am facing is `timerReadMicros` always return older time when it was last called, and if I call it second time it seems to be close to real value. It looks like `timerReadMicros` is storing and returning older values. I have tried `timerRead` , `timerReadSeconds` . Every function have same behavior. Following is the code snippet

```cpp
void IRAM_ATTR entry_isr()
{
    value = timerReadMicros(timer); // will always return value it was called the last time
}
```

if I change the code like this

```cpp
void IRAM_ATTR entry_isr()
{
    timerReadMicros(timer); // will always return value it was called the last time
    value = timerReadMicros(timer); // kind of hack to get correct value
}
```

timer configuration is like this

```cpp
  timer = timerBegin(0,40,true); //Input clock 80MHz, Prescaler = 40, Upward counting ==> 0.5microsecond per counter increment
  timerWrite(timer, 0); //reset timer to 0
```

I assume I am missing something very basic here.

Any help is really appreciated.

atanisoft, 2020-07-12:
duplicate of https://github.com/espressif/arduino-esp32/issues/3434. There needs to be a small delay inside timerRead(), you can try `delayMicroseconds(1);` instead of `delay(1);` which was mentioned in the other issue (a 1ms delay inside the ISR *WILL* cause problems).

atanisoft, 2020-07-12:
also if you just need an incremental timestamp you can call `esp_timer_get_time();` which will return the number of microseconds since boot.

ummarbhutta, 2020-07-13:
Hi @atanisoft, I wouldn't go for updating the library code and adding delay after update. Not sure why a delay is required, isn't it preventing us from full control on hardware timers? e.g. If I am using timerRead with scaling less than 80 than I can get resolution of less than 1 microseconds.. But putting delay will make it hard.

Anyways thanks for second suggestion, for now I read resolution in microseconds so will give it a try.

closed 2020-09-26 by stale[bot]