espressif/arduino-esp32 #4162: Add basic functions to library files

alve89, 2020-07-13:
Hey guys,

I have a general question (only with the example SD library, I don't know the status with the other libraries).

Why don't you put the basic functions (written in the example [SD_Test](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino)) in the [library files](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.cpp) itself? Using a SD card will (almost) everytime desire to use these functions.
Therefore it was more efficient to implement them in one file rather than let every user "redefine" it in the own projects.

Regards!

isadora-6th, 2020-07-15:
I think during testing like stuff like SD_Test you usually want to edit code, because sometimes you have problem in resistor, capacitor, or in your main project something strange happens when SD mixed with WiFi or similar.

Also, as I know, examples attached with lib files, in ```examples``` folder, so you similarly copy them.
Also hiding stuff like SD_Test -> basic and very simple code in source code is a problem.

Because instead of simple example you have  to read ```src``` files what is hard when you only start your way, or want to test something.
Also don't you will have project with single line in setup ```SD_test_example();```? No profit putting examples inside, maybe in comment section it can be usefull but in code...

alve89, 2020-07-16:
Either I understand you wrong or there is a misunderstanding. From my point of view a functions like `readFile()` or `newFile()` could be quite useful as default functions directly within the library. So if a user wants to create a new file he just needs to run `newFile()` and not `SD.fileOpen()` and `SD.fileClose()` (or something like that).

lbernstone, 2020-07-16:
This project is a programming framework, not a disk operating system.  The goal here is to provide an Arduino API for the esp32.  Perhaps something like [micropython](http://micropython.org/) is more what you are looking for?

closed 2020-09-29 by stale[bot]