espressif/arduino-esp32 #4163: HTTPS POST <body> is lost during redirect ESP2866/Arduino

GamerClassN7, 2020-07-13:
Hi,
i am working on custome SimpleHome (SmartHome Server) project i run in to the issue. If i want to use HTTPs, ESP lose request payload i realy try to do every possible alternative and nothing worked can someone sugest what i am dooing wrong ? i am running code on ESP2866 custome borad based on NodeMCU. Thanks for any answer.

Server response
Make your question, not a Statement, inclusive. Include all pertinent information:

### Description:
ESP is not able to send POST body on redirect, but if i test API with Insomnia of Curl everithing works fine.
### Server Response:
[image]

### Sketch:  https://raw.githubusercontent.com/Simple-Home/SmartHome-Firmwares/master/firmwares/PlatformIO/S20/src/main.cpp

lbernstone, 2020-07-13:
esp32 != esp8266
https://github.com/esp8266/Arduino/issues

GamerClassN7, 2020-07-15:
@lbernstone Thanks!

closed 2020-07-15 by GamerClassN7