espressif/arduino-esp32 #4164: ESP 32 (Lolin32 v1.0.0) deep sleep still powers 5v pin

georgepopovici, 2020-07-13:
####################

Hardware:
Board: Lolin32 v1.0.0
Core Installation version: 1.0.0
IDE name: Arduino IDE 1.8.13
Flash Frequency: 80Mhz
PSRAM enabled: ?
Upload Speed: 115200
Computer OS: Windows 10
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Auto-detected Flash size: 4MB

Description:
Hello,

When using deep sleep, I see that the my GPS NEO module is still having power (the 5V pin). I would expect it to shut off when deep sleep starts.

Is this the expected behavior?

George

Sketch: (leave the backquotes for code formatting)
```cpp
//Change the code below by your sketch
#include "ThingSpeak.h"
#include "secrets.h"
#include <WiFi.h>
#include <SoftwareSerial.h>
#include <TinyGPS.h>
#include "DHT.h"
#include "ESP32_MailClient.h"

#define DHTPIN 32
#define DHTTYPE DHT22
#define emailSenderAccount    "XXX"    
#define emailSenderPassword   "XXX"
#define emailRecipient        "XXX"
#define smtpServer            "smtp.gmail.com"
#define smtpServerPort        465
#define emailSubject          "Message from caravan"

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  15        /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

/* This sample code demonstrates the normal use of a TinyGPS object.
   It requires the use of SoftwareSerial, and assumes that you have a
   4800-baud serial GPS device hooked up on pins 4(rx) and 3(tx).
   GPS TX - LOLIN port 33
   GPS RX - LOLIN port 34
*/

DHT dht(DHTPIN, DHTTYPE);
TinyGPS gps;
SoftwareSerial ss(33, 34);
SMTPData smtpData;
WiFiClient  client;

char ssid[] = SECRET_SSID;   // your network SSID (name) 
char pass[] = SECRET_PASS;   // your network password
int keyIndex = 0;            // your network key Index number (needed only for WEP)

unsigned long myChannelNumber = SECRET_CH_ID;
const char * myWriteAPIKey = SECRET_WRITE_APIKEY;
const char * myReadAPIKey = SECRET_READ_APIKEY;

// Initialize our values
String myStatus = "";

void sendCallback(SendStatus info);

void setup() {
  Serial.begin(115200);  //Initialize serial
  delay(500);

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");

  WiFi.mode(WIFI_STA);   
  ThingSpeak.begin(client);  // Initialize ThingSpeak
  ss.begin(9600);
  dht.begin();
  
  Serial.print("Simple TinyGPS library v. "); Serial.println(TinyGPS::library_version());
  Serial.println("by Mikal Hart");
  Serial.println();

  int statusCode = 0;
  // Connect or reconnect to WiFi
  if(WiFi.status() != WL_CONNECTED){
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(SECRET_SSID);
    while(WiFi.status() != WL_CONNECTED){
      WiFi.begin(ssid, pass);  // Connect to WPA/WPA2 network. Change this line if using open or WEP network
      Serial.print(".");
      delay(5000);     
    } 
    Serial.println("\nConnected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // Get GPS data
  bool newData = false;
  unsigned long chars;
  unsigned short sentences, failed;

  // For one second we parse GPS data and report some key values
  for (unsigned long start = millis(); millis() - start < 1000;)
  {
    while (ss.available())
    {
      char c = ss.read();
      // Serial.write(c); // uncomment this line if you want to see the GPS data flowing
      if (gps.encode(c)) // Did a new valid sentence come in?
        newData = true;
    }
  }

  if (newData)
  {
    float flat, flon;
    unsigned long age;
    gps.f_get_position(&flat, &flon, &age);
    Serial.print("LAT=");
    Serial.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flat, 6);
    Serial.print(" LON=");
    Serial.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flon, 6);
    Serial.print(" SAT=");
    Serial.print(gps.satellites() == TinyGPS::GPS_INVALID_SATELLITES ? 0 : gps.satellites());
    Serial.print(" PREC=");
    Serial.print(gps.hdop() == TinyGPS::GPS_INVALID_HDOP ? 0 : gps.hdop());
    ThingSpeak.setField(3, flat);
    ThingSpeak.setLatitude(flat);
    ThingSpeak.setField(4, flon);
    ThingSpeak.setLongitude(flon);
  }
  
  gps.stats(&chars, &sentences, &failed);
  Serial.print(" CHARS=");
  Serial.print(chars);
  Serial.print(" SENTENCES=");
  Serial.print(sentences);
  Serial.print(" CSUM ERR=");
  Serial.println(failed);
  if (chars == 0)
    Serial.println("** No characters received from GPS: check wiring **");

  //Get DHT22 data
  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);

  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println(F("Failed to read from DHT sensor!"));
    return;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print(F("Humidity: "));
  Serial.print(h);
  Serial.print(F("%  Temperature: "));
  Serial.print(t);
  Serial.print(F("°C "));
  Serial.print(f);
  Serial.print(F("°F  Heat index: "));
  Serial.print(hic);
  Serial.print(F("°C "));
  Serial.print(hif);
  Serial.println(F("°F"));

  // set the fields with the values
  // 1 is humidity
  // 2 is temperature
  // 3 and 4 are latitude and longitude
  
  ThingSpeak.setField(1, h);
  ThingSpeak.setField(2, t);
  ThingSpeak.setField(7, bootCount);

  ThingSpeak.setStatus(myStatus);
  
  // write to the ThingSpeak channel
  int x = ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey);
  if(x == 200){
    Serial.println("Channel update successful.");
  }
  else{
    Serial.println("Problem updating channel. HTTP error code " + String(x));
  }
  long h_thres = ThingSpeak.readLongField(myChannelNumber, 8, myReadAPIKey);  

   // Check the status of the read operation to see if it was successful
  statusCode = ThingSpeak.getLastReadStatus();
  if(statusCode == 200){
    Serial.println("H_thresh: " + String(h_thres));
    if(h > h_thres){
      //send email
      smtpData.setLogin(smtpServer, smtpServerPort, emailSenderAccount, emailSenderPassword);
      smtpData.setSender("ESP32", emailSenderAccount);
      smtpData.setPriority("High");
      smtpData.setSubject(emailSubject);
      smtpData.setMessage("Humidity is: "+ String(h) + " - Expected < " + String(h_thres), false);
      smtpData.addRecipient(emailRecipient);
      smtpData.setSendCallback(sendCallback);
      if (!MailClient.sendMail(smtpData))
        Serial.println("Error sending Email, " + MailClient.smtpErrorReason());
      smtpData.empty();
    }
      
  }
  else{
    Serial.println("Problem reading channel. HTTP error code " + String(statusCode)); 
  }
  
  Serial.println("Going to sleep now");
  delay(1000);
  Serial.flush(); 
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
  //delay(20000); // Wait 20 seconds to update the channel again

}

void loop() {
  
}

void sendCallback(SendStatus msg) {
  // Print the current status
  Serial.println(msg.info());

  // Do something when complete
  if (msg.success()) {
    Serial.println("----------------");
  }
}
```

lbernstone, 2020-07-13:
The 5V line is the source power (presumably from the computer's usb in your case).  The esp32 has no control over it.  You could set up a transistor to control the power into the gps device if that is what you want.

georgepopovici, 2020-07-13:
Thanks @lbernstone !
What I would want is to have as less power as possible consumed while in deep sleep. But I see the GPS still working.

The GPS module could be powered by the 3V ports also, according to its specs. Are the 3V ports expected to power off while deeps sleep?

Thanks,
George

lbernstone, 2020-07-13:
You can control which power domains are active during deep sleep, but that is not the issue.  Maximum power from a esp32 pin is 42mA, and really you should not be powering other devices through the chip (this generates heat in the chip).  Learn how to set up a transistor to control power flow.

georgepopovici, 2020-07-14:
Thanks @lbernstone . I'll see what I can do.

All the best,
George

closed 2020-07-14 by georgepopovici