espressif/arduino-esp32 #4167: ESPAsyncWebServer & WiFiClient (for HTTPUpdate)

g01d10x, 2020-07-14:
Hello everyone,

I'm trying to find an example for my ESP32 to be connected to a WiFi network with access to the internet, and running a functioning ESPAsyncWebServer. BUT .. I would also like to have the ESP32 also connect to perform an HTTPUpdate with a .bin stored on my server.

I've tried the first HTTPUpdate example from [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPUpdate/examples) by itself and all works well, but when I add the same code into my project with the ESPAsyncWebServer, the WDT gets triggered from the async_tcp on CPU 1.

I've tried several implementations to have the server and a separate client coexist without a crash, but no luck. I am afraid the only way I might get this to happen is to have the whole program restart, then perform the HTTPUpdate process before the server.begin(), but this seems a little extreme so I'm hoping for help.

Otherwise the ESPAsyncWebServer from @me-no-dev seems to be a solid server for the rest of everyone's needs, but I can't figure out how to get it to play nicely when there's another client alive which is connecting out to the internet.

I've also tried adding all of these 4 lines before starting up the HTTPUpdate process in various combinations to no avail.
```
server.end();
disableCore0WDT();
disableCore1WDT();
disableLoopWDT();
```
Any help would be appreciated !

Backtrace info below:
```
E (31237) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (31237) task_wdt:  - async_tcp (CPU 1)
E (31237) task_wdt: Tasks currently running:
E (31237) task_wdt: CPU 0: IDLE0
E (31237) task_wdt: CPU 1: async_tcp
E (31237) task_wdt: Aborting.
abort() was called at PC 0x400ebca3 on core 0

Backtrace: 0x4008c800:0x3ffbe170 0x4008ca31:0x3ffbe190 0x400ebca3:0x3ffbe1b0 0x40084b3d:0x3ffbe1d0 0x40168f8f:0x3ffbc2f0 0x400ed05a:0x3ffbc310 0x4008a72d:0x3ffbc330 0x40088f49:0x3ffbc350
  #0  0x4008c800:0x3ffbe170 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008ca31:0x3ffbe190 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400ebca3:0x3ffbe1b0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40084b3d:0x3ffbe1d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40168f8f:0x3ffbc2f0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x400ed05a:0x3ffbc310 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008a72d:0x3ffbc330 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x40088f49:0x3ffbc350 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```
Here's my hardware information
```
Board:					DOIT ESP32 Devkit V1
Core Installation version:	Espressif 32 1.12.4
IDE name:				PlatformIO & VS Code (Mac)
Flash Frequency:			?40Mhz?

PLATFORM: Espressif 32 1.12.4 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)

RAM:   [=         ]  13.3% (used 43528 bytes from 327680 bytes)
Flash: [=======   ]  71.6% (used 938422 bytes from 1310720 bytes)
Chip is ESP32D0WDQ6 (revision 1)
Changing baud rate to 460800
```

Computer OS:                                        Mac OSX

lbernstone, 2020-07-14:
This is not the appropriate forum for this question, as it is not an issue with the arduino-esp32 code.  Ask on esp32.com

BlueAndi, 2020-07-21:
Move this issue to the ESPAsyncWebserver github repository. Just as a note, you need too long in the http request, thats why the tadk wd is triggered.

closed 2020-10-04 by stale[bot]