espressif/arduino-esp32 #4168: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.   (about mqtt)

masdel123, 2020-07-14:
### Hardware:
Board:							wemos lolin 32
Core Installation/update date:			?11/jul/2017?
IDE name:							Platform io
Flash Frequency:					80
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            windows 10

### Description:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
When I want to receive battery voltage and passing it to mqtt server , I get an error about below the debug messages

I want to pass battery voltage to mqtt server .

### Debug Messages:
```
ttempting to connect to WPA SSID: -GT-
192.168.1.101

Async Scanning sample starting

About to start scanning forever seconds
MQTT reconnect...Found an EddystoneTLM beacon!

>>>>>>>>>>>>>>>>>>>>>>> Device Added 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2e0  PS      : 0x00060030  A0      : 0x800d22c8  A1      : 0x3ffe6270  
A2      : 0x3ffe6324  A3      : 0x00000000  A4      : 0x00000013  A5      : 0x3ffe6324  
A6      : 0x00ff0000  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffe6240  
A10     : 0x3ffe62c8  A11     : 0x3f403852  A12     : 0x00000001  A13     : 0x3ffe62b4  
A14     : 0x00000030  A15     : 0x000000f3  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4000c2e0:0x3ffe6270 0x400d22c5:0x3ffe6280 0x400d1c0e:0x3ffe62a0 0x400d3809:0x3ffe6360 0x400d2ab2:0x3ffe6450 0x4013cc29:0x3ffe64a0 0x40137d9a:0x3ffe64c0 0x4008e711:0x3ffe64f0

PC: 0x4000c2e0
EXCVADDR: 0x00000000

Decoding stack results
0x400d1d9d: BLEAdvertisedDevice::getServiceUUID() at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEAdvertisedDevice.cpp line 128
0x400d181e: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at C:\Users\-GT-\Desktop\mqtt ve batarya\elf_i_in/elf_i_in.ino line 212
0x400d42fd: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEScan.cpp line 119
0x400d3076: BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEDevice.cpp line 270
0x4013b7a9: btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c line 54
0x4013691a: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008e711: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
0x400d1d9d: BLEAdvertisedDevice::getServiceUUID() at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEAdvertisedDevice.cpp line 128
0x400d181e: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at C:\Users\-GT-\Desktop\mqtt ve batarya\elf_i_in/elf_i_in.ino line 212
0x400d42fd: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEScan.cpp line 119
0x400d3076: BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\-GT-\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE\src\BLEDevice.cpp line 270
0x4013b7a9: btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c line 54
0x4013691a: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008e711: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2020-07-14:
[Exception decoder](https://github.com/me-no-dev/EspExceptionDecoder)

masdel123, 2020-07-14:
I have already added decoder code...

lbernstone, 2020-07-14:
Doh.  Didn't see the scroll bars.  Your error indicates a problem with BLE.  We are not mind readers.  You need to provide minimal code which reproduces the problem and pin down what the error is.

masdel123, 2020-07-14:
here is my code

```
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
	void onResult(BLEAdvertisedDevice advertisedDevice)
	{
		if (wifiConnected)
		{
			rssi = advertisedDevice.getRSSI(); 		//get the RSSI Value
			battery = advertisedDevice.getServiceDataUUID().toString().c_str();

			
		uint8_t *payLoad = advertisedDevice.getPayload();
BLEUUID checkUrlUUID = (uint16_t)0xfeaa;

if (advertisedDevice.getServiceUUID().equals(checkUrlUUID))
		{
			if (payLoad[11] == 0x10)
			{
				// this part for Eddystone URL beacon
			}
			else if (payLoad[11] == 0x20)
			{
				Serial.println("Found an EddystoneTLM beacon!");
				BLEEddystoneTLM foundEddyURL = BLEEddystoneTLM();

				// Why The  is a std::string used here. THE ADVERTISED DATA IS BINARY AND CONTAINS NULL VALUES
				std::string eddyContent = "01234567890123";

				for (int idx = 0; idx < 14; idx++)
				{
					eddyContent[idx] = payLoad[idx + 11];
				}

				foundEddyURL.setData(eddyContent);
				String batarya=String(foundEddyURL.getVolt());
			/*	Serial.printf("Reported battery voltage: %dmV\n", foundEddyURL.getVolt());
				Serial.printf("Reported temperature from TLM class: %.2fC\n", (double)foundEddyURL.getTemp());
				// Correct calculation of temperature from 8.8 fixed-point
				int temp = (int)payLoad[16] + (int)(payLoad[15] << 8);
				float calcTemp = temp / 256.0f;
				Serial.printf("Reported temperature from data: %.2fC\n", calcTemp);
				Serial.printf("Reported advertise count: %d\n", foundEddyURL.getCount());
				Serial.printf("Reported time since last reboot: %dms\n", foundEddyURL.getTime());
				Serial.println("\n");
*/

				strcpy(beaconAddress, advertisedDevice.getAddress().toString().c_str()); // get the Beacon MAC
			//Loop with the AddressesCount as the counter and check if the Device is detected before
			 //If so append its RSSI Value to the Related Row in the beaconsPackets
			for (int i = 0; i <= AddressesCount; i++)
			{
				if (strncmp(beaconAddress, beaconsPackets[i], 15) == 0)
				{
					deviceFound = true;
					itoa(rssi, rssiString, 10);
					strcat(beaconsPackets[i], " ,");
					strcat(beaconsPackets[i], rssiString);
					if (strlen(beaconsPackets[i]) > 63)
					{
						//printf("Founded Beacons : %d\n", AddressesCount);
						strcat(beaconsPackets[i], theReaderName);
					String msg2=	strcat(beaconsPackets[i], ",");

					msg2.concat(batarya);

						checkAndSend(beaconsPackets[i]);
						printf("\n>>>>>>>>>>>>>>>>>>>>>> Packet Sent \n");
						printf(beaconsPackets[i]);

						

							strcpy(beaconsPackets[i], beaconAddress);

					
						mqtt.beginPublish(topic.c_str(), msg2.length(), false);
						mqtt.print(msg2.c_str());
						
					}
					break;
				}
				else
				{
					deviceFound = false;
				}
			}
		}

			
			}
			if (!deviceFound)
			{
				printf("\n>>>>>>>>>>>>>>>>>>>>>>> Device Added \n");
				strcpy(beaconsPackets[AddressesCount], beaconAddress);
				AddressesCount++;
			}
		}
	}
};

```

chegewara, 2020-07-14:
it looks like this crash, now check why (add logs):
`strcpy(beaconsPackets[AddressesCount], beaconAddress);`

masdel123, 2020-07-14:
strcpy(beaconsPackets[i], beaconAddress);

this code is true because this runs before added the battery voltage code.but when I added battery codes for receiving battery voltage it didnt run...

chegewara, 2020-07-14:
Thats all i can help now.

lbernstone, 2020-07-14:
So it works if you change `String batarya=String(foundEddyURL.getVolt());` to `String batarya="3.3";`?

masdel123, 2020-07-14:
this code is true??

```
void setup()
{
	periodicReboot = timerBegin(0, 80, true);
	 timerAlarmWrite(periodicReboot, 300000000, false);
	timerAttachInterrupt(periodicReboot, &interruptReboot, true);
	timerAlarmEnable(periodicReboot);
	loopTime = millis();
 // pinMode (ledPin, OUTPUT); //led uygulama
  // pinMode (ledPin2, OUTPUT); //led uygulama
	//fire a Watchdog timer on Core0 with a interrupt function
	//to fix the issue of hanging the code when the default wdt triggered
	xTaskCreatePinnedToCore(
		Core0WDT, /* Function to implement the task */
		"Core0WatchdogTimer", /* Name of the task */
		10000,  /* Stack size in words */
		NULL,  /* Task input parameter */
		0,  /* Priority of the task */
		&Core0WdtHandle,  /* Task handle. */
		0); /* Core where the task should run */

	Serial.begin(115200);
	status = WiFi.begin(ssid, pass);
	
  WiFi.softAP(ssid, pass); 
	// attempt to connect to Wifi network:

	
	mqtt.setServer(MQTT_SERVER, MQTT_PORT);
	
  mqtt.setCallback(mqtt_receive_callback);
  topic = MQTT_PREFIX_TOPIC + my_mac + MQTT_BLE_TOPIC;
  delay(1000);
	connectWifi();
	//connectServer();
	
	
}

void loop()
{

	
	printf("\nAsync Scanning sample starting\n");
	BLEDevice::init("");
	BLEScan *pBLEScan = BLEDevice::getScan();
	pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
	pBLEScan->setActiveScan(true);
	printf("\nAbout to start scanning forever seconds\n");
	pBLEScan->start(0, scanCompleteCB);
	//
	// you can consider this loop as the main Thread
	//
	while (1)
	{
		checkMqtt();
	mqtt.loop();
		checkConnection();
		printf("\nTick! - still alive");
		FreeRTOS::sleep(1000);
	}
}
```

lbernstone, 2020-07-14:
This is not a code review site.  YOU ARE RESPONSIBLE FOR FINDING YOUR ERROR.

masdel123, 2020-07-14:
I know this!

masdel123, 2020-07-14:
I only learn where is my mistake in my program about this post topic

MarkyAD, 2020-07-15:
@masdel123 The problem is in your thinking that is is a helpdesk. This is an issue tracker for reporting issues with the core code. I believe the others have established that the problem lies in your code. Help is at esp32.com, not here.

masdel123, 2020-07-15:
ok. I understand.

closed 2020-07-16 by masdel123