espressif/arduino-esp32 #4169: free() target pointer is outside heap areas

nnangeroni, 2020-07-14:
### Hardware:
Board:							ESP32-Pico-D4 rev 1, 240 MHz, 4MB Flash
Core version:                                           none
IDE name:						Arduino IDE 1.8.13
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I am trying to read the battery level from a BLE temperature sensor.  
I've used the BLE_client example code with some changes to MyAdvertisedDeviceCallbacks() in order to identify the target device.  
Occasionally it succeeds and produces the expected result.
Most of the time it crashes immediately after finding the desired characteristic.  
Any idea what's causing the crashes?
Thanks,
Nancy

### backtrace decode:

Decoding stack results
0x40091448: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40091679: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400e9acb: _strtod_r at ../../../.././newlib/libc/stdlib/strtod.c line 192
0x40084b1f: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x400850e1: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x400d5cb2: BLERemoteCharacteristic::retrieveDescriptors() at c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h line 544
0x400d69a9: BLERemoteService::getCharacteristic(BLEUUID) at C:\Users\User\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteService.cpp line 151
0x400d49f1: BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at C:\Users\User\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEDescriptor.cpp line 151
0x400d5751: BLERemoteCharacteristic::readValue[abi:cxx11]() at C:\Users\User\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteCharacteristic.cpp line 419
0x401073e5: btc_av_state_started_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/a2dp/btc_av.c line 734
0x40100486: _btc_storage_add_ble_bonding_key at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_ble_storage.c line 113
0x4008e089: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "BLEDevice.h"
static BLEUUID serviceUUID("180F");
static BLEUUID    charUUID("2A19");
 
static boolean foundSensor = false;
static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }
  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
    Serial.print("Connecting... ");    
[... 204 lines cut ...]
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: 00002a19-0000-1000-8000-00805f9b34fb, handle: 27 0x1b
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe3ba8), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe3ba8), owner: readValue
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x400e9acb on core 0

Backtrace: 0x40091448:0x3ffcf3a0 0x40091679:0x3ffcf3c0 0x400e9acb:0x3ffcf3e0 0x40084b1f:0x3ffcf410 0x400850e1:0x3ffcf430 0x4000bec7:0x3ffcf450 0x400d5cb2:0x3ffcf470 0x400d69a9:0x3ffcf4b0 0x400d49f1:0x3ffcf4d0 0x400d5751:0x3ffcf580 0x401073e5:0x3ffcf5f0 0x40100486:0x3ffcf640 0x4008e089:0x3ffcf670

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...

```

lbernstone, 2020-07-15:
You should check to be sure value has a length >1 or else you are reading past the end of the string.

nnangeroni, 2020-07-15:
Thanks, lbernstone, will do.

closed 2020-09-28 by stale[bot]