espressif/arduino-esp32 #4170: Control Lolin32 which of the 3V pins has power

georgepopovici, 2020-07-15:
####################

Hardware:
Board: Lolin32 v1.0.0
Core Installation version: 1.0.0
IDE name: Arduino IDE 1.8.13
Flash Frequency: 80Mhz
PSRAM enabled: ?
Upload Speed: 115200
Computer OS: Windows 10
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Auto-detected Flash size: 4MB

Description:
Hello,

I would like to control which of the 3V pins has power or not. Is this possible just from Arduino code?

The use case is that I have humidity sensor and GPS module connected and when ESP32 goes to deep sleep, they still get power. I'd like to shut them off when ESP32 is in deep sleep and power on when ESP wake up. I have another noise sensor that is connected to another 3V pin. I'd like that to have power via deep sleep since it is waking up the board when noise detected.

Thanks,
George

sansillusion, 2020-07-15:
You need to use a transistor to control power to external devices.

georgepopovici, 2020-07-16:
Thanks @sansillusion . Would you please be able to recommend one suitable transistor?
Regards,
George

sansillusion, 2020-07-17:
A quick google gave me that page with a good discussion about using MOSFETs
(low current consumption transitors)

https://forum.allaboutcircuits.com/threads/looking-for-a-mosfet-to-use-with-attiny-running-on-3v-2xaaa.151970/

georgepopovici, 2020-07-17:
I've ordered this one NDP6020P. Will try it and add also results here.

Thanks,
George

closed 2020-09-30 by stale[bot]