espressif/arduino-esp32 #4171: Changing Serial baudrate causes InstrFetchProhibited when Udp.parsePacket() called

isadora-6th, 2020-07-15:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation version:			1.0.4(platform.io) (also tested 1.0.2 arduino)
IDE name:						Arduino IDE , Platform.io
Flash Frequency:					240Mhz
PSRAM enabled:                                     Disabled
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I'm sending Art-Net UDP packets from MADRIX to ESP32 controller.   
ESP32 there is Acess Point with static IP.
When I call ```Serial.begin()``` twice => on ```Udp.receivePacket()``` I get Exception ```(InstrFetchProhibited )```

Also I noticed, that debug messages not appear when begin called twice.

Twice or more during tests i recived message    
```<TIME_THERE> addba response cb: ap bss deleted```
before exception

Please note, when using ESP32 in STA mode even if I called ```Serial.begin()``` twice, packets still received. 
P.S. Exception appears when packet sent from my PC to controller IP

```cpp

#include <Arduino.h>
#include <WiFi.h>

WiFiUDP Udp;

void setup() {

  Serial.begin(9600); //
  /*Even if Serial.end() called problem still there*/
  //Serial.end();
  Serial.begin(115200);

  WiFi.softAPConfig(IPAddress(192,168,4,99),IPAddress(192,168,4,1),IPAddress(255,255,255,0));
  WiFi.softAP("NETWORK","Secured4897");
  
  Udp.begin(6454);
  Serial.println("Setup_finish");
}

uint32_t simple_timer_to_prevent_serial_spam = 0;
void loop() {
  uint32_t packLen = Udp.parsePacket();
  if(packLen > 0){
    Udp.flush();
    if(millis() - simple_timer_to_prevent_serial_spam > 300){
      simple_timer_to_prevent_serial_spam = millis();
      Serial.println("packLen="+String(packLen));
    }
  }  
}
```

### Debug Messages:
<details>
  <summary>Expand Exception</summary>
  

```java
Backtrace: 0x00000000:0x3ffb3d30 0x400ec36e:0x3ffb3d70 0x400f75dd:0x3ffb3d90 0x400faf71:0x3ffb3dd0 0x401001a6:0x3ffb3df0 0x400ef15b:0x3ffb3e10 0x4008877d:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400ec36e: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x400f75dd: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x400faf71: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x401001a6: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x400ef15b: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```
</details>    

<details>
  <summary>Expand Decoded Exception</summary>
  

```cpp
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400ec36e: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x400f75dd: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x400faf71: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x401001a6: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x400ef15b: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```
</details>    

<details>
  <summary>Serial port when begin() called once</summary>
  

```cpp
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
Setup_finish
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 16 - AP_STADISCONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 18 - AP_PROBEREQRECVED
packLen=530
packLen=530
packLen=530
packLen=530
packLen=530

```
</details>    

</details>    

<details>
  <summary>Serial port when begin() called twice (not matter with Serial.end() or without)</summary>
  

Also I noticed, that debug messages not appear when begin called twice.
Even if it build flag in ```platform.io``` or menu option in ```Arduino```
```cpp
Setup_finish
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x8010f991  A1      : 0x3ffb3d30  
A2      : 0x3ffcc8e8  A3      : 0x3ffcdec0  A4      : 0x3ffcbe28  A5      : 0x3ffcbe08  
A6      : 0x6404a8c0  A7      : 0x6e04a8c0  A8      : 0x8010f834  A9      : 0x3ffb3cf0  
A10     : 0x3ffcc8f8  A11     : 0x3ffcdec0  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x4010f98e:0x3ffb3d70 0x4011b401:0x3ffb3d90 0x40120409:0x3ffb3dd0 0x401256a6:0x3ffb3df0 0x4011022b:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Setup_finish

```
</details>

isadora-6th, 2020-07-16:
```Serial.updateBaudRate(int);``` Works for me, but any way, calling ```Serial.end()``` causes exception.
Should I write to issue list in ```lwip``` section? I think problem is in ```Serial.end()``` code section, some internal resources (ap bss? Don't know what it is) corrupted.

isadora-6th, 2020-07-16:
During work with ```SD_MMC``` in one bit mode also noticed, not using ```Serial.end()```(using updateBaudRate) fixed not opening files on ```SD.open(F_name);``` (I fixed it using ```while(time+100ms){ SD.open(File)}``` but delays and error messages still was sent.
Something deeper there.

lbernstone, 2020-07-16:
There is a fix for Serial.end() issues. https://github.com/espressif/arduino-esp32/pull/3894

closed 2020-09-29 by stale[bot]