espressif/arduino-esp32 #4177: Connect to device with BLE Pin code

nathan-casabieille, 2020-07-17:
Hello, This is my first question, I hope it is the right place to ask it.

I want to get data from a device, but it has a BLE Pin code (123456).

I managed to connect on the device but I can neither read nor write in the characteristics.

I did some research on the internet. I found server side codes, but no client side codes.
Some people even said that it is not yet possible, but these subjects date from a few years ago.

If someone could enlighten me on this subject it would be really nice, because it's been a long time since I got stuck on this problem.

Thank You !

chegewara, 2020-07-17:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_passkey/BLE_client_passkey.ino#L21

nathan-casabieille, 2020-07-17:
Thank you very much @chegewara, that's exactly what I needed !

closed 2020-07-17 by nathan-casabieille