espressif/arduino-esp32 #4181: stuck at rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2)) waiting for download

dnkhal, 2020-07-18:
Greetings community,

I faced the following issue, at some point the wroom 32 boards stucks at 
rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2)) 
waiting for download

I know that this states is caused when the GPIO0 is pulled to the ground but I didn't do that in a first case and secondly there is no available GPIO0 pin on the board itself, board like this
[image]

Using Arduino IDE but convinced that the problem may be in the HW cause there is no problem to write a FW but to evaluate its functionality I need to go out of download mode

So any ideas on how to approach such issue?

dnkhal, 2020-07-18:
> It sounds like either the BOOT button which is tied to IO0 is faulty or there is a faulty / missing RTS/DTS signal handling circuit on the PCB.

I;ve checked the button and it is working properly (no shorts), what can I do about RTS/DTS?
Thanks

atanisoft, 2020-07-18:
Likely not much other than get a replacement board.

closed 2020-07-18 by dnkhal

chegewara, 2020-07-18:
If you can then try to measure voltage on pin0 and EN.
If its broken then pin0 should be LOW, about 0.7V or less. EN probably too.

wang-xianwen, 2023-04-03:
I have also had this problem recently, and the chip has been OK several times before. Suddenly, the chip was unable to run the program and remained in download mode. Have you found a solution yet

chegewara, 2023-04-03:
> I have also had this problem recently, and the chip has been OK several times before. Suddenly, the chip was unable to run the program and remained in download mode. Have you found a solution yet

And what did you try to do so far?

wang-xianwen, 2023-04-03:
> > I have also had this problem recently, and the chip has been OK several times before. Suddenly, the chip was unable to run the program and remained in download mode. Have you found a solution yet
> 
> And what did you try to do so far?

The level of the IO0 pin I measured has always been 3.3V, but it is unclear why the chip cannot enter flash. Do you have any ideas

chegewara, 2023-04-03:
From this point of view its impossible, so it has to be some severe damage of the board.