espressif/arduino-esp32 #4183: Exception was unhandled. Memory dump at 0x400ef984: ffffffff ffffffff ffffffff

aminSharifi21, 2020-07-18:
### Hardware:
Board:							ESP32 Devkit
Core Installation/update date:			18/7/2020
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 8.1

### Description:

i try to read data from Temperature and humidity sensor and CO
sensor and CO sensor  than show on LCD with I2C protocol, but i have this ERROR. 

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ef984: ffffffff ffffffff ffffffff
Core 1 register dump:
PC      : 0x400ef988  PS      : 0x00060130  A0      : 0x800d5066  A1      : 0x3ffb1f80  
A2      : 0xffffffff  A3      : 0x3ffc0568  A4      : 0x00000020  A5      : 0x80000020  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x800d14de  A9      : 0x3ffb1f60  
A10     : 0xffffffff  A11     : 0x00000000  A12     : 0x00000016  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400ef988:0x3ffb1f80 0x400d5063:0x3ffb1fb0 0x4008b2e1:0x3ffb1fd0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

### Debug Messages:

PC: 0x400ec79c
EXCVADDR: 0x00000000

lbernstone, 2020-07-18:
https://github.com/espressif/arduino-esp32#decoding-exceptions

aminSharifi21, 2020-07-19:
> 
> 
> https://github.com/espressif/arduino-esp32#decoding-exceptions

[image]

closed 2020-10-02 by stale[bot]