espressif/arduino-esp32 #4184: How to compile esp-idf to use with arduino

RdlP, 2020-07-18:
### Hardware:
Board:							ESP32-WROVER-B
IDE name:						Arduino IDE?
Flash Frequency:					40Mhz
Upload Speed:						?115200?
Computer OS:                                            Ubuntu 18.04

### Description:
I want to modify some parts of the esp BluetoothLibrary (libbt.a) to use it with arduino (especially to remove the SSP profile in order to establish a fixed pin). However, I can't figure out how I can compile the esp-idf project to(to generate libbt.a).

So, my question is how can I compite esp-idf and get libbt.a?

Thanks.

lbernstone, 2020-07-18:
Things are currently stretched a bit, but the standard method is [lib-builder](https://github.com/espressif/esp32-arduino-lib-builder/issues/12).  Please read through issues to compile ATM.

RdlP, 2020-07-19:
Thanks @lbernstone for your quickly response.

Finally, following this https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md and this https://github.com/espressif/arduino-esp32/issues/3760 I have compiled the libraries for arduino. 

But now I hava a new problem. I copy the following files:

 - /home/user/esp/esp32-arduino-lib-builder/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.cpp
 - /home/user/esp/esp32-arduino-lib-builder/components/arduino/libraries/BluetoothSerial/src/BluetoothSerial.h
 - /home/user/esp/esp32-arduino-lib-builder/out/tools/sdk/lib/libbt.a

Into its corresponding folders:

 - /root/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/BluetoothSerial/src for BluetoothSerial.cpp and BluetoothSerial.h
 - /root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib for libbt.a

The problem is that now, in arduino, when I try to compile SerialToSerialBT example i get the following errors:

```
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib/libbt.a(bt.o):(.literal.esp_bt_controller_init+0x2c): undefined reference to `coex_ble_adv_priority_high_set'
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib/libbt.a(bt.o): In function `esp_bt_controller_init':
/home/user/esp/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1510: undefined reference to `coex_ble_adv_priority_high_set'
libraries/BluetoothSerial/BluetoothSerial.cpp.o:(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x1c): undefined reference to `esp_bt_gap_ssp_confirm_reply'
libraries/BluetoothSerial/BluetoothSerial.cpp.o:(.literal._ZN15BluetoothSerial5beginE6Stringb+0x20): undefined reference to `esp_bt_gap_set_security_param'
libraries/BluetoothSerial/BluetoothSerial.cpp.o: In function `esp_bt_gap_cb(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*)':
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/WString.h:48: undefined reference to `esp_bt_gap_ssp_confirm_reply'
libraries/BluetoothSerial/BluetoothSerial.cpp.o: In function `BluetoothSerial::begin(String, bool)':
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/WString.h:48: undefined reference to `esp_bt_gap_set_security_param'
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib/libbt.a(bta_dm_main.o):(.literal.BTA_DmCoexEventTrigger+0x0): undefined reference to `esp_coex_status_bit_set'
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib/libbt.a(bta_dm_main.o):(.literal.BTA_DmCoexEventTrigger+0x4): undefined reference to `esp_coex_status_bit_clear'
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib/libbt.a(bta_dm_main.o): In function `BTA_DmCoexEventTrigger':
/home/user/esp/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_main.c:469: undefined reference to `esp_coex_status_bit_set'
/home/user/esp/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_main.c:470: undefined reference to `esp_coex_status_bit_clear'
/home/user/esp/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_main.c:473: undefined reference to `esp_coex_status_bit_clear'
/home/user/esp/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_main.c:474: undefined reference to `esp_coex_status_bit_clear'
collect2: error: ld returned 1 exit status
exit status 1
Compiling error for ESP32 Wrover Module board.
```

Any advice to solve these undefined references?
Thanks.

lbernstone, 2020-07-19:
`IDF_COMMIT=4e27cbb89 ./build.sh`

RdlP, 2020-07-19:
Thanks @lbernstone. After some tricks I have achieved to get it work!.

I have several questions in order to entirely understand it.

First of all, why must I use `IDF_COMMIT=4e27cbb89 ./build.sh`? I think that commit is the v3.3 but why I should use that version?

The second question is that after I executed `IDF_COMMIT=4e27cbb89 ./build.sh` and copied the files into its corresponding folders, I got the following undefined references in arduino:

```
libraries/BluetoothSerial/BluetoothSerial.cpp.o:(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x10): undefined reference to `esp_bt_gap_ssp_confirm_reply'
libraries/BluetoothSerial/BluetoothSerial.cpp.o:(.literal._ZN15BluetoothSerial5beginE6Stringb+0x18): undefined reference to `esp_bt_gap_set_security_param'
libraries/BluetoothSerial/BluetoothSerial.cpp.o: In function `esp_bt_gap_cb(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*)':
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.h:104: undefined reference to `esp_bt_gap_ssp_confirm_reply'
libraries/BluetoothSerial/BluetoothSerial.cpp.o: In function `BluetoothSerial::begin(String, bool)':
/root/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Stream.h:104: undefined reference to `esp_bt_gap_set_security_param'
collect2: error: ld returned 1 exit status
```

Before I build the libbt.a, I executed `make menuconfig` to disable `SSP`. So, I think that these errors are related with SSP functions that do not exist since I compiled libbt.a without SSP support (it is correct?). But I expected that this was handle in BluetoothSerial.cpp (Why it is not handle?). In order to fix it I have had to comment the calls to those functions in BluetoothSerial.cpp

Thank you 
Regards,

lbernstone, 2020-07-20:
If your changes are not confined to libbt.a, you will need to copy all the libraries that are changed in order to have a consistent set.  It is typically easiest to just copy the whole thing with tools/copy-to-arduino.sh

GianCann, 2020-07-22:
Hi guys,
I want to do the same thing, but using the library on Arduino for Windows.
I used WSL (Windows Subsytem for Linux) to rebuild the library...
I need to things:
- Modified the max BLE connections (from 3 to 8), using make menuconfig
- Modified BLEServer.cpp/BLEServer.h to add a method to Callback class

Now, I able to compile all, anche in /arduino-esp32-lib-builder/out/ I have all recompile library.

And now? 
Where should I put these files on Windows?
I did various tests but without success...

Pallanez, 2020-07-23:
> Hi guys,
> I want to do the same thing, but using the library on Arduino for Windows.
> I used WSL (Windows Subsytem for Linux) to rebuild the library...
> I need to things:
> 
> Modify of the maximum connections for BLE (from 3 to 8), using make menuconfig
> Modify BLEServer.cpp/BLEServer.h to add a method to Callback class
> Now, I able to compile all, and in /arduino-esp32-lib-builder/out/ I have all recompiled library.
> 
> And now?
> Where should I put these files on Windows?
> I did various tests but without success...

I think you'd have to add them to your_arduino_esp32_framework_folder\tools\sdk\lib.

Is there any way to recompile arduino libraries on Windows without the need for a subsystem for Linux? I can't get arduino-esp32-lib-builder to work on my Windows machine.

GianCann, 2020-07-24:
@Pallanez yes, I find where to place the file but i get various errors from the linker.

> Is there any way to recompile arduino libraries on Windows without the need for a subsystem for Linux?

I want to try to setup a enviroment based on Git Bash, on Windows.

Pallanez, 2020-07-24:
@GianCann What errors are you getting from the linker? Did you git clone the master branch of the arduino repo? I was able to compile on a virtual Ubuntu 14.04 machine applying @fededim 's patches (https://github.com/espressif/esp32-arduino-lib-builder/issues/12).

Voha888, 2020-09-20:
> Thanks @lbernstone. After some tricks I have achieved to get it work!.

Hello, do You have **libbt.a** for **framework-arduinoespressif32 3.10004.200129 (1.0.4)** with disabled **SSP**? Please upload here what You have

closed 2021-02-04 by stale[bot]