espressif/arduino-esp32 #4185: How to use ESP NOW local master key?

somehibs, 2020-07-18:
ESP NOW offers set_pmk but doesn't offer a set_lmk. How do I set my lmk?

Master: set_pmk: 123 _set_lmk_: 555
Peer: set_pmk, add_peer(lmk: 555)

ESP NOW doesn't have set_lmk, how does the master know what his lmk is?

lbernstone, 2020-07-18:
No idea, but doco is at [esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html)

somehibs, 2020-07-19:
Yeah, I read the documentation. It was clear that you need to set an LMK for the PMK to be used during transmission, but it doesn't mention how to configure an LMK for the receiving device, so how does the receiving device decrypt the LMK packet? Both devices share a PMK, but only one knows the LMK.

somehibs, 2020-07-20:
I've tried adding the receiver as a peer on both devices, with the LMK, MAC, channel and encrypt boolean set identically. The PMK for both devices is identical.

The sender states the message was sent successfully in the send callback, but the receiver does not decrypt or pass the message on to the recv callback.

lbernstone, 2020-07-20:
CSB. These are all esp-idf functions.  You should submit this issue at [esp-idf](https://github.com/espressif/esp-idf/issues) or take it to esp32.com for discussion.

somehibs, 2020-07-23:
The issue is that the Arduino environment doesn't offer those methods.

somehibs, 2020-07-23:
Well, whatever.

closed 2020-07-23 by somehibs