espressif/arduino-esp32 #4186: BLE_notify_example_crash, when Core debug level was not set to "Verbose"

marcrupprath, 2020-07-19:
Hello,
i try to use BLE_NOTIFY example and have dicovered that is will crash on notify activation 
if core debug info is not set into "Verbose mode"

Once Notifiy is activated in NRF connect i got plenty of messages in logger like:
09:26:14.363 -> load:0x40080400,len:5816
09:26:14.363 -> entry 0x400806ac
09:26:15.357 -> [I][BLEDevice.cpp:554] getAdvertising(): create advertising
09:26:15.357 -> Waiting a client connection to notify...
09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.208 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.208 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
09:26:34.208 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error

In case that coredebug level is set to "verbose" in Arduino IDE all works fine, wonder why

Greatings Marc

marcrupprath, 2020-07-19:
And addition: On connect with "nrf connect app and verbose mode active i got ongoing messages like:

V][GeneralUtils.cpp:323] hexDump(): 0000 5f 00 00 00                                      _...<CR><LF>
[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring<CR><LF>
[D][BLECharacteristic.cpp:794] onStatus(): BLECharacteristicCallbacks<CR><LF>
[D][BLECharacteristic.cpp:795] onStatus(): BLECharacteristicCallbacks<CR><LF>
[V][BLECharacteristic.cpp:646] setValue(): >> setValue: length=4, data=60000000, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8<CR><LF>
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffddf2c), owner: <N/A> for <Unknown><CR><LF>
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffddf2c), owner: <Unknown><CR><LF>
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SetValue (0x3ffddf2c), owner: <Unknown><CR><LF>
[V][BLECharacteristic.cpp:655] setValue(): << setValue<CR><LF>
[V][BLECharacteristic.cpp:477] notify(): >> notify: length: 4<CR><LF>
[D][BLECharacteristic.cpp:782] onNotify(): BLECharacteristicCallbacks<CR><LF>
[D][BLECharacteristic.cpp:783] onNotify(): BLECharacteristicCallbacks<CR><LF>
[V][GeneralUtils.cpp:296] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f<CR><LF>
[V][GeneralUtils.cpp:297] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --<CR><LF>
[V][GeneralUtils.cpp:323] hexDump(): 0000 60 00 00 00                                      `...<CR><LF>
[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring<CR><LF>
[D][BLECharacteristic.cpp:794] onStatus(): BLECharacteristicCallbacks<CR><LF>
[D][BLECharacteristic.cpp:795] onStatus(): BLECharacteristicCallbacks<CR><LF>
[V][BLECharacteristic.cpp:646] setValue(): >> setValue: length=4, data=61000000, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8<CR><LF>
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffddf2c), owner: <N/A> for <Unknown><CR><LF>
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffddf2c), owner: <Unknown><CR><LF>
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SetValue (0x3ffddf2c), owner: <Unknown><CR><LF>
[V][BLECharacteristic.cpp:655] setValue(): << setValue<CR><LF>
[V][BLECharacteristic.cpp:477] notify(): >> notify: length: 4<CR><LF>
[D][BLECharacteristic.cpp:782] onNotify(): BLECharacteristicCallbacks<CR><LF>
[D][BLECharacteristic.cpp:783] onNotify(): BLECharacteristicCallbacks<CR><LF>
[V][GeneralUtils.cpp:296] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f<CR><LF>
[V][GeneralUtils.cpp:297] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --<CR><LF>
[V][GeneralUtils.cpp:323] hexDump(): 0000 61 00 00 00                                      a...<CR><LF>
[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring<CR><LF>

chegewara, 2020-07-19:
Here you are sending notifications when no client is connected yet:
```
09:26:15.357 -> Waiting a client connection to notify...
09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
```

Not sending notifications, because client didnt write value 1 to CCC descriptor:
`[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring`
`[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring`

marcrupprath, 2020-07-19:
That does not answr the questions, why application crashes, if debug mode was not set to "verbose".
Do you have tested it and can explain it why is it ?

chegewara, 2020-07-19:
It is tested in examples. Besides of examples i have done dozens ble apps.

marcrupprath, 2020-07-19:
Hello, did not answer the question at all:
1. 09:26:34.173 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
Waht may it casue , it happens once you activate notification. and Verbose mode was not set as debug level

2. What may be the reason why it is working when egug level is thet to "verbose" ?

Question is why core debug level has such an influence ?

chegewara, 2020-07-19:
Its not problem with debug level, its problem with your test:
1) this log is printed only in verbose:
`[V][BLECharacteristic.cpp:498] notify(): << notifications disabled; ignoring`
and 2 lines later function returns
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.cpp#L498-L500

2) this log is printed here:
`[E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error`
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.cpp#L524

> Question is why core debug level has such an influence ?

Answer: its not debug log, its your testing procedure is not the same in both cases

marcrupprath, 2020-07-19:
"its your testing procedure is not the same in both cases"
I try to explain again;

I open up the "ESP_notify.ino" Sketch inside my Arduino as example and left the code untouched.
My Board is ESP32 Thing plus, i tested with ESP Thing as well as Arduino ESP feater as board slected , same beaviour:

a) I use Core debug Level " verbose" --> all fine
b) use Core debig Level "none" for example , causes crash if notifiy is activated.

Where is my difference in testing here ? Did not understand, please explain again

Thanks

chegewara, 2020-07-19:
And where is part of log that shows it crashed?
Did you try to use this tool to check why it crashed:
https://github.com/me-no-dev/EspExceptionDecoder

Maybe its the problem with lack of initialization for one parameter and that line returns true or false because of garbage:
`p2902 != nullptr && !p2902->getNotifications()`

You have word `crash` in title and description but nothing show when, where or why it crashed. Maybe someone with more patient will try to help you, because im done.

marcrupprath, 2020-07-19:
"Crash" means that Nordic cnrf connect has canceled connection
I will try to investigate it according to your hints above in more detail
Nice weekend

closed 2020-07-19 by marcrupprath

marcrupprath, 2020-07-19:
Solution was as follows:
1.
Use an  Object: Example:
BLECharacteristic LEGOCharacteristic(CHARACTERISTIC_UUID, BLECharacteristic::PROPERTY_READ | ......);

2. Add Characteristics to server characteristics collection
 pService->addCharacteristic(&LEGOCharacteristic);

Note : My special variation - on usage of BLE2902 Descriptor 
....
BLE2902 Test = BLE2902();
BLEDescriptor *MyDesciptor  =  &Test;
......
3. Assign Descritor to this Characteristics object :   LEGOCharacteristic.addDescriptor(&Test);
....

4. Write Values and Notify
.....
           LEGOCharacteristic.setValue(........);
            LEGOCharacteristic.notify();

Thanks for all help and pushing my nose in right direction ;-)

chegewara, 2020-07-19:
I just told you what is wrong, you should listen, not arguing.
Whithout code i could only tell where you should look at.