espressif/arduino-esp32 #4187: Conflict Between Wifi & analogRead() in ESP32s

marwinwijaya, 2020-07-19:
[image]
I have a problem about esp32s, where when I use the Wifi.h lib some analogRead () always produces a value of 4095, I try on pin 0 of esp32s, is there a pin that can be used simultaneously between wifi and analogRead ()? I attach a picture pinout esp32s Thanks you

sansillusion, 2020-07-19:
If I remember correctly, ADC2 pins are not capable of analogread when wifi
is in use. Use an ADC1 pin instead.

marwinwijaya, 2020-07-19:
ADC1 pin is IO34, IO35, IO32, IO33 @sansillusion ?

sansillusion, 2020-07-19:
I would think so, I did not have to use ADC yet, just remember seeing this being a recurrent issue for people here.

atanisoft, 2020-07-19:
@marwinwijaya all pins above 30 are ADC1.

marwinwijaya, 2020-07-20:
thank you @sansillusion and @atanisoft

closed 2020-10-03 by stale[bot]

qiaoyukeji, 2023-07-09:
我在用 esp32cam 时，wifi 与 adc2 不能同时使用，使用 wifi 时 adc2 读数一直为0。后来，我尝试交替使用 wifi 与 adc2 ，例如每10秒钟断开wifi一次，读取adc2的值，然后重新连接wifi，使两个功能都能使用，经过测试，能够正常使用。

Arduino 的代码如下：
```
#include <Arduino.h>
#include <WiFi.h>
#include "esp_camera.h"
#include <vector>
#include <string.h>
using namespace std;

// ph and tem 变量设置
const int phPin = 2;
float ph;
float Value = 0;

// wifi 
const char *ssid = "TP-LINK_1760";
const char *password = "987654321";

void wifi_init() {
  WiFi.mode(WIFI_STA);
  WiFi.setSleep(false);  //关闭STA模式下wifi休眠，提高响应速度
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi Connected!");
  Serial.print("IP Address:");
  Serial.println(WiFi.localIP());
}

void setup() {

  Serial.begin(115200);
  //  wifi 初始化
  wifi_init();

  pinMode(phPin, INPUT);
}
void loop() {
  // 关闭 wifi
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
  
  // adc2 读取 phPin 上的数据
  pinMode(phPin, INPUT);
  Value = analogRead(phPin);
  
  // 打开 wifi
  wifi_init();
  Serial.print(Value);
  delay(500);
}
```