espressif/arduino-esp32 #4191: Are SD operations thread safe?

curly-brace, 2020-07-20:
I'm interested whether SD operations like opening a file or creating a directory safe to be called without additional mutex from different threads running on different cores.

I see that is it based on VFS and i've found the documentation on VFS, but there is nothing written about thread safety, except two functions that set line endings for UART, which makes me wonder if everything else in VFS is thread safe or not.

lbernstone, 2020-07-20:
The examples you have given are stateless, so don't need thread safety.  Writing to a file from two threads will not be thread safe with SD (or FS in general).  There is no flock in xtensa, so you must set a semaphore to get exclusive access.

curly-brace, 2020-07-20:
I was not clear, sorry. I don't write to the same file, but there may be situation while one thread writes to one file and another thread decides to read another file. Mainly I concerned here about SPI bus lock, not a file lock.

lbernstone, 2020-07-20:
I don't know the raw details, but there is an IO scheduler.  If not, this would cause crashes, not thread safety issues.

closed 2020-10-03 by stale[bot]