espressif/arduino-esp32 #4195: Faster SPI library?

DFL1973, 2020-07-22:
Is the SPI library limited to 26Mhz? It hangs if I use something greater then that.

`     hspi = new SPIClass(HSPI);
      hspi->begin();
      hspi->setFrequency(40000000L) ; //works with 26000000L
`
I'm using the correct pins for the HSPI (GPIO 12,13,14,15) and the supplied example "SPI_Multiple_Buses" as a guide.

The Espressif Programming Manual we can read "While in general, speeds up to 80MHz on the dedicated SPI pins and 40MHz on GPIO-matrix-routed pins are supported, full-duplex transfers routed over the GPIO matrix only support speeds up to 26MHz.". On other section " To retain a higher frequency, you have to use the IO_MUX pins or the dummy bit workaround. ". Finally, "If the Host only writes data, the dummy bit workaround and the frequency check can be disabled by setting the bit SPI_DEVICE_NO_DUMMY in the member spi_device_interface_config_t::flags. When disabled, the output frequency can be 80MHz, even if the GPIO matrix is used."

Well, I'm guessing I'm using the "IO_MUX" pins, since it is the default, anyway, I'm not finding an obvious way to set the "dummy" flag, so, my questions is: 

Is it possible to use the maximum speed on the SPI? Can someone provide a working example?
Or is it a bug or limitation from the Arduino port?

lbernstone, 2020-07-22:
The arduino driver is written to use the gpio matrix.  You will need to use the idf functions if you wish to use the direct io_mux.
https://github.com/espressif/esp-idf/tree/v3.3.2/examples/peripherals/spi_master

DFL1973, 2020-07-22:
Thanks! Very helpful!

I'm trying to get a minimum setup just for the tests, but I'm having some issues

`
#include <Arduino.h>
#include "driver/spi_master.h"

#define SPI_NUM  HSPI
#define PIN_NUM_MOSI 13
#define PIN_NUM_MISO 12 
#define PIN_NUM_SCK  14
#define PIN_NUM_CS   15 

spi_device_handle_t FPGAspi;

    
void initSPI()
{
   esp_err_t err;
    spi_bus_config_t  bus_config;
    spi_host_device_t host;
    spi_device_interface_config_t dev_config;

       bus_config.mosi_io_num = PIN_NUM_MOSI;
      bus_config.miso_io_num = PIN_NUM_MISO;
      bus_config.sclk_io_num = PIN_NUM_SCK;
      bus_config.quadwp_io_num = -1;
      bus_config.quadhd_io_num = -1;
      bus_config.intr_flags = 0;        // NOTE 1
    
    dev_config.mode = 0;
    dev_config.clock_speed_hz = 20000000;  
    dev_config.spics_io_num = PIN_NUM_CS;   
    dev_config.queue_size = 1;  

    err=spi_bus_initialize(HSPI_HOST, &bus_config, 1);
    ESP_ERROR_CHECK(err);
    err = spi_bus_add_device(HSPI_HOST, &dev_config, &FPGAspi);
    ESP_ERROR_CHECK(err);
}

void setup (void)
{
      Serial.begin(115200);
      while (!Serial) 
      {
          ; // wait for serial port to connect. Needed for native USB port only
      }
    
      Serial.println("Started...");
      

      initSPI();
}  

void loop (void)
{
}`

With the above code, I got: 

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
Started...
E (34) spi: spicommon_bus_initialize_io(186): mosi not valid
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008cff0
file: "SPI_speed_test.ino" line 38
func: void initSPI()
expression: err

Backtrace: 0x4008cb10:0x3ffb3ed0 0x4008cff3:0x3ffb3ef0 0x400d0e02:0x3ffb3f10 0x400d0e35:0x3ffb3f80 0x400d1f2b:0x3ffb3fb0 0x400897c5:0x3ffb3fd0

Rebooting...

NOTE 1: Without this line I got
E (34) spi_master: spi_bus_initialize(236): intr flag not allowed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008cff0

with the line:
E (34) spi: both wp and hd required.
E (35) spi: spicommon_bus_initialize_io(213): not all required capabilities satisfied.
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008cff0

Maybe a wrong "intr_flag"?

lbernstone, 2020-07-22:
You probably need to hard cast some of those config variables to type.

closed 2020-10-04 by stale[bot]