espressif/arduino-esp32 #4196: HTTPClient.h not compatible with TinyGsm

bgondell, 2020-07-22:
### Hardware:
Board:							ESP32 Dev Module v1
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:

**Summary: Code works fine on WiFi, but on GPRS it crash. Given examples works fine also.**
Modem: SIM800L
TinyGSM version: 0.10.5 (latest on Library Manager)

Note: _First versions of code used ArduinoHTTPClient.h, which actually worked on Wifi/GPRS, but entire code was very different (not only the part on using Wifi or GPRS). Then I've changed to HTTPClient.h because code had others problems related to data, not communication itself._

### Code: < related code because full code has  900 lines >

```cpp
...
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER 1024
#define SerialAT Serial2
#define SerialMon Serial
#include <HTTPClient.h>

#define USE_GPRS true
#define USE_WIFI false

#ifdef USE_GPRS
  #include <TinyGsmClient.h>
#else
  #include <WiFi.h>
#endif

#ifdef USE_GPRS
    #ifdef DUMP_AT_COMMANDS
      #include <StreamDebugger.h>
      StreamDebugger debugger(SerialAT, SerialMon);
      TinyGsm modem(debugger);
    #else
      TinyGsm modem(SerialAT);
    #endif
  TinyGsmClient client(modem);
#else
  WiFiClient wifi;
#endif

const char apn[]  = "virtueyes.vivo.com.br";
const char gprsUser[] = "vivo";
const char gprsPass[] = "vivo";
const char* url = "http://XXX.XXX.XXX.XXX/ws/saveData";

void setup() {
...

  if ( USE_GPRS ) {
    SerialAT.begin(38400); // Set GSM module baud rate.
    ConnectGPRS(1); // 0 = modem init. 1 = modem restart. Pode levar algunms minutos.
  };

  if ( USE_WIFI ) {
    Serial.println("Connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    
    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.print(".");
[... 151 lines cut ...]

> Decoding stack results
> 0x4008c50c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
> 0x4008c73d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
> 0x4008890d: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1442
> 0x401217a2: sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 78
> 0x401219a2: sys_arch_protect at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 469
> 0x40113e04: do_memp_malloc_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 302
> 0x40113e6d: memp_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 398
> 0x4011100a: netconn_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c line 742
> 0x4010fe94: netconn_new_with_proto_and_callback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c line 133
> 0x40113688: lwip_socket at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 1587
> 0x400d759e: WiFiClient::connect(IPAddress, unsigned short, int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/sockets.h line 593
> 0x400d6e7a: WiFiClient::connect(char const*, unsigned short, int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 281
> 0x400d683e: HTTPClient::connect() at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 1017
> 0x400d6921: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 552
> 0x400d69f3: HTTPClient::POST(unsigned char*, unsigned int) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 493
> 0x400d6a13: HTTPClient::POST(String) at C:\Users\evelis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 498
> 0x400d491d: TaskCommunication(void*) at C:\Users\evelis\Documents\Arduino\PortaV2.3/PortaV2.3.ino line 234
> 0x40088c21: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
> 

### Debug and AT command log : 

> Now the circuit is under review by the electronic technician to check if there are faults on that side. When the circuit board is reviewed, I will debug if necessary.

Jeroen88, 2020-07-24:
HTTPClient will only work with a WiFiClient. You might try my PR #3848, that handles the HTTP protocol by directly manipulating the Client passed into the constructor, however I never tried it with TinyGsm.

bgondell, 2020-07-24:
Thanks @Jeroen88 , I'll give it a try!

bgondell, 2020-07-31:
Hi @Jeroen88 , at first try isn't working. I'll debug a little more and share the results.

It's just a post so if someone know another solution to be share, will be welcome!

Jeroen88, 2020-08-01:
Sim800 has a http client also using AT commands. If I recall correctly tinygsm itself may have a http client as well (that uses these AT commands). If I recall correctly Sim800 only supports http,no https, so the server you're connecting to should allow unsecured http connections

bgondell, 2020-08-05:
@Jeroen88 Exactly, I'm using an HTTP server to manage POST.

First I made a code to test HTTP GET with your library. Success Rate is bad, changing setTimeout (up to 100000) appears not to affect on nothing, error -1 comes just a couple of seconds later.

> 23:23:01.336 -> TRYING HTTP GET FIRTS TO CHECK  CONNECTION
> 23:23:03.529 -> Request failed with error code -1
> 23:23:03.562 -> Closing connection
> 23:23:03.562 -> Success: 14. Intents: 81. Success Rate: 17%

Now... HTTP POST has a more acceptable success rate around 50%. (Also with setTimeout at 100000, and getting -1 error a couple of seconds later).

> 00:07:01.598 -> NOW TESTING TO MAKE A POST
> 00:07:03.898 -> Request failed with error code -1
> 00:07:03.936 -> Closing connection
> 00:07:03.936 -> Success: 95. Intents: 194. Success Rate: 48%

Maybe a fuction to retry connection? or maybe wait a bit more?

Thanks!!

Jeroen88, 2020-08-05:
I have no access to my computer now, so I can not check anything before next week.
Could you add debug statements in the library itself to drill down the exact point of failure?

I understand correctly that sometimes it works? That means that in principle it is possible to make http requests over a Sim800 connection.

closed 2020-10-18 by stale[bot]

mazorvanunu, 2020-10-28:
excuse me but why is this stale exactly? can the esp32 httpclient support tinygsmclient or any other than wificlient?
why, on god lovely green earth, will the httpclient talk to anything other than the stream(tinygsm/wifi "client")? and most certainly of all, why on earth will it talk directly with the sim800 via at commands?????????????? i mean wow!

OptimumAssociatesUK, 2022-08-11:
Hi bgondell! Did you resolve the issue and got success to send data via GSM?

bgondell, 2022-08-24:
I used <ArduinoHttpClient.h> instead <HTTPClient.h>, split the data in 1024-bits, send the 1024-bit parts and finally the last part is sended at last in another code because it can be less than 1024-bits.

My very very ugly but working code:

`      if ((data.length() <= 1024) || USE_WIFI) {
        Serial.println("Sending one package");
        http.post(resource, contentType, data);
      }
      else {  // split and sent 1024-bit parts
        uint8_t temp[1025];
        int chunks = (data.length() / 1024);
        Serial.print("Big packet, spliting in ");Serial.print(chunks);Serial.println(" parts.");
        http.startRequest(resource, "POST", "application/json", data.length());
        for (int i = 0; i < chunks; i++) {
          data.substring( i * 1024 , ((i + 1) * 1024) ).getBytes(temp, 1025);
          Serial.println((char*)temp);
          http.write(temp, 1024);
          delay(10);
        }
        int lastpiece = data.length() - (chunks * 1024) + 1;
        if (lastpiece > 0) {
          uint8_t temp2[lastpiece];
          data.substring(chunks * 1024).getBytes(temp2, lastpiece); //Last piece of data
          Serial.println((char*)temp2);
          http.write(temp2, lastpiece);
        }`

OptimumAssociatesUK, 2022-08-25:
Thank you very much for your answer budy!
Your solution is a very good one; congrats!

I found a very nice solution that worked very well using ArduinoJson.h and
TinyGsmClient.h
Basically this is the code:

 //JSON doc generation and serialization

                StaticJsonDocument<384> doc;

                doc["api_key"] = apiKeyValue;
                doc["sensor"] = "Reflections";
                doc["value1"] = (parsedStrings[0]);
                doc["value2"] = (parsedStrings[1]);
                doc["value3"] = (parsedStrings[2]);
                doc["value4"] = (parsedStrings[3]);
                doc["value5"] = (parsedStrings[4]);
                doc["value6"] = (parsedStrings[5]);
                doc["value7"] = (parsedStrings[6]);
                doc["value8"] = (parsedStrings[7]);
                doc["value9"] = (parsedStrings[8]);
                doc["value10"] = (parsedStrings[9]);
                doc["value11"] = (parsedStrings[10]);

                serializeJson(doc, jsonOutput);

                // Prepare your HTTP POST request data (Temperature in
Celsius degrees)
                String httpRequestData = jsonOutput;

                //JSON Transmission 2 the server
                client.print(String("POST ") + resource + " HTTP/1.1\r\n");
                client.print(String("Host: ") + server + "\r\n");
                client.println("Connection: close");
                client.println("Content-Type: application/json");
                client.print("Content-Length: ");
                client.println(httpRequestData.length());
                client.println();
                client.println(httpRequestData);

                unsigned long timeout = millis();
                while (client.connected() && millis() - timeout < 10000L) {
                  // Print available data (HTTP response from server)
                  while (client.available()) {
                    char c = client.read();
                    SerialMon.print(c);
                    timeout = millis();
                  }
                }
                SerialMon.println();

                // Close client and disconnect
                client.stop();