espressif/arduino-esp32 #4199: ESP32 CANBUS messages repeating continuously.

yogeshwaran774, 2020-07-24:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE

### Description:
HI i am using CANBUS in ESP32 for such a long time and i made some modifications such BUS-OFF will get recovered, but after sometime one of my ESP keeps receiving same messages continuously. This issue persist even if i disconnect other ESP s from bus and connect back without switching off the affected ESP which is receiving messages.

If i hard reset the ESP then it starts working properly. What could be the issue?

atanisoft, 2020-07-24:
without code it will be impossible to assist. This should also likely be reported over on esp32.com or https://github.com/espressif/esp-idf since there is no arduino wrapper provided by this repository.

yogeshwaran774, 2020-07-24:
> without code it will be impossible to assist. This should also likely be reported over on esp32.com or https://github.com/espressif/esp-idf since there is no arduino wrapper provided by this repository.

Good to see you back again. I feel either there can be issue with CAN trans receiver or CAN controller present inside the esp32 because the same code is working fine with other node which are connected to the same bus. Only one node has this issue and it dose not have terminating resistor.

I can explain the problem more detail. There were Four ESP32 devices(A,B,C,D) connected with CANBUS with each spaced around 2.5m length. Power source is same for all and when switch on first time all the devices works perfectly, then after sometime device **'A'** keeps receiving the same message even though other devices sends proper message. I checked serial prints from other devices there everything was perfect. This issue happens only with device A not with other devices.

**Library**
https://github.com/sandeepmistry/arduino-CAN

**Code snippet** 
void setup{
 CAN.setPins(2, 4); 
  if (!CAN.begin(125E3)) {
    Serial.println("Starting CAN failed!");
  }
xTaskCreatePinnedToCore(can_handle_task, "can_handle_msg", 5000, NULL, 3, &task_2, 0);
}
void loop()
{
  CAN.beginPacket(random(3,10));
 byte c_data[8];
memset(c_data,0x09,8);
  for (int i = 0; i < 8; i++)
  {
    CAN.write(c_data[i]);
  }
  CAN.endPacket();
  delay(1000);
  }
void can_handle_task(void * param)  //C@H
{
 for(;;)
{
  int packetSize = CAN.parsePacket();
if(packetSize)
{
  Serial.println(CAN.packetId());
}
vTaskDelay(50 / portTICK_PERIOD_MS);
}
}

atanisoft, 2020-07-24:
This should likely be routed to https://github.com/sandeepmistry/arduino-CAN since you are using that library. But that said, I would suggest switching which pins you are using for CAN to 4 and 5 (these are the recommended defaults from IDF).

yogeshwaran774, 2020-07-24:
> This should likely be routed to https://github.com/sandeepmistry/arduino-CAN since you are using that library. But that said, I would suggest switching which pins you are using for CAN to 4 and 5 (these are the recommended defaults from IDF

okay i will change the pins as you said and verify. Let you know the results.

closed 2020-10-07 by stale[bot]

michaelHH1, 2021-01-29:
> okay i will change the pins as you said and verify. Let you know the results.

@yogeshwaran774: did you ever solved that issue? I have the same problem and would highly appreciate any hint. Thx a lot in advance

ingsoc, 2021-03-04:
@michaelHH1 - I had the same issue.  I had better luck with https://github.com/miwagner/ESP32-Arduino-CAN.

I modified one of the examples to just write anything received on the CAN BUS (I was connecting to a vehicle via the OBD2 port)

You will need to alter the r/tx pins and Serial BAUD rate.

```
#include <Arduino.h>
#include <ESP32CAN.h>
#include <CAN_config.h>

CAN_device_t CAN_cfg;               // CAN Config
unsigned long previousMillis = 0;   // will store last time a CAN Message was send
const int interval = 1000;          // interval at which send CAN Messages (milliseconds)
const int rx_queue_size = 10;       // Receive Queue size

void setup() {
  Serial.begin(115200);
  Serial.println("Basic Demo - ESP32-Arduino-CAN");
  CAN_cfg.speed = CAN_SPEED_500KBPS; // <--------------------
  CAN_cfg.tx_pin_id = GPIO_NUM_32;   // <--------------------
  CAN_cfg.rx_pin_id = GPIO_NUM_33;   // <--------------------
  CAN_cfg.rx_queue = xQueueCreate(rx_queue_size, sizeof(CAN_frame_t));
  // Init CAN Module
  ESP32Can.CANInit();
}

void loop() {

  CAN_frame_t rx_frame;

//   unsigned long currentMillis = millis();

  // Receive next CAN frame from queue
  if (xQueueReceive(CAN_cfg.rx_queue, &rx_frame, 3 * portTICK_PERIOD_MS) == pdTRUE) {

    if (rx_frame.FIR.B.FF == CAN_frame_std) {
      printf("New standard frame");
    }
    else {
      printf("New extended frame");
    }

    if (rx_frame.FIR.B.RTR == CAN_RTR) {
      printf(" RTR from 0x%08X, DLC %d\r\n", rx_frame.MsgID,  rx_frame.FIR.B.DLC);
    }
    else {
      printf(" from 0x%08X, DLC %d, Data ", rx_frame.MsgID,  rx_frame.FIR.B.DLC);
      for (int i = 0; i < rx_frame.FIR.B.DLC; i++) {
        printf("0x%02X ", rx_frame.data.u8[i]);
      }
      printf("\n");
    }
  }
}
```