espressif/arduino-esp32 #4202: ESP-32, mcp23017, rotaryEncoderOverMCP23017 endless restart when using A6 at 23017

maxnemoy, 2020-07-25:
I am using the library https://github.com/maxgerhardt/rotary-encoder-over-mcp23017

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.12.4
IDE name:						Platform.io
Upload Speed:					115200
Computer OS:                                          Windows 10.2004

### Description:
I get endless reboot. The error occurs when initializing pin A6, at 23017

for example

```
RotaryEncOverMCP(&mcp, 6, 14, &RotaryEncoderChanged, 1);
```
for other pins it works well.

This problem is relevant only for esp32. 
(the same circuit is tested for ATMEGA)

I understand that maybe my question is at the wrong address. But the author abandoned the library. I hope for any advice that can help solve the problem. I tried a lot, but I don't have a debugger for esp32, so I can't look into the chip to find the problem.

If you cannot help, just close this. thank.

### Sketch
you can use the example of this library for testing
https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Polling/PollingExample.cpp

### Debug Messages:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40080fd9  PS      : 0x00060330  A0      : 0x800d0dab  A1      : 0x3ffe3b70
A2      : 0x0000000e  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x3f405fa4  A9      : 0xffffffff
A10     : 0xffffffff  A11     : 0x00000030  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0xffffffff  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffffa  

Backtrace: 0x40080fd9:0x3ffe3b70 0x400d0da8:0x3ffe3b90 0x400d0f0d:0x3ffe3bb0 0x400d4fb3:0x3ffe3bd0 0x400820f9:0x3ffe3bf0 0x400822d5:0x3ffe3c20 0x40079053:0x3ffe3c40 0x400790b9:0x3ffe3c70 0x400790c4:0x3ffe3ca0 0x4007928d:0x3ffe3cc0 0x400806de:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
```

lbernstone, 2020-07-25:
I can't help with this particular library, but you can decode the error message with the [backtrace decoder](https://github.com/espressif/arduino-esp32#decoding-exceptions).  Also, if the 6 in that line is a pin reference, that is a reserved pin. See the [pinmap](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap)

atanisoft, 2020-07-25:
> Also, if the 6 in that line is a pin reference, that is a reserved pin.

From a very quick scan of the referenced library it is a pin on the MCP23017 and not the ESP32

atanisoft, 2020-07-25:
*HOWEVER* [these](https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/lib/Rotary-master/Rotary.cpp#L73-L83) lines will cause issues since it is randomly resetting pin modes when the [constructor](https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/src/RotaryEncOverMCP.h#L28) runs.

maxnemoy, 2020-07-25:
@lbernstone  Thank. I will try this.

@atanisoft MCP23017 works great on its own, any pins. It seems to me that the problem is exactly here https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/lib/Rotary-master/Rotary.cpp

but it is not clear to me why this works well for arduino, but does not work for esp.

I'm new to this, maybe my question is stupid.
Thanks for the help, I will try to fix this.

atanisoft, 2020-07-25:
> but it is not clear to me why this works well for arduino, but does not work for esp.

Because arduino (avr) does not restrict usage of certain exposed pins. The ESP32 has a handful of restricted pins that are connected to the on-board flash which can not be used by the user.

maxnemoy, 2020-07-25:
> > but it is not clear to me why this works well for arduino, but does not work for esp.
> 
> Because arduino (avr) does not restrict usage of certain exposed pins. The ESP32 has a handful of restricted pins that are connected to the on-board flash which can not be used by the user.

But I don’t use pin esp, 6pin for 23017.
if we take esp32 and 23017, the Adafruit_MCP23017 library everything works fine, including reading and writing to A6 23017.

But for this particular case, using the Rotary library, everything breaks and I get an error.

my english is not that good. maybe I'm not explaining well.

still many thanks. you can close this. i hope i will solve it

atanisoft, 2020-07-25:
> But I don’t use pin esp, 6pin for 23017.

You aren't using it directly but the Rotary library *IS* as I linked above to the constructor. It blindly resets the pin mode and pull-up values for whatever pins you are sending into the RotaryEncOverMCP declaration. I'm honestly not entirely sure how this library doesn't break applications on the AVR side as well since it will randomly reset pin modes!

maxnemoy, 2020-07-25:
@lbernstone I am very sorry, I understood what you were talking about.
I managed to fix it.

Thanks again so much for the tip.

closed 2020-07-25 by maxnemoy