espressif/arduino-esp32 #4207: How can I install the ESP32-S2 support in Arduino?

ziv2013, 2020-07-26:
I have brought a ESP32-S2-Saola-1 from Espressif. And I want to try it with Arduino IDE(in fact I want to try the USB function of S2). But I find I can't find this board in Board Manager.  I have tried "Installation Instructions" in the main page, but there is no support for S2.

My question is how can I add S2 support in Arduino? 

Thanks in advance.

chegewara, 2020-07-26:
ESP32 S2 support is in esp32s2 branch.
You have to delete old esp32 installation, if it was done with board manager, and use instructions from `Using Arduino IDE with the development repository`, then switch branch.

ziv2013, 2020-07-27:
Thanks a lot. I have enabled my USBTiny on ESP32 with your help.

closed 2020-07-27 by ziv2013

jukeeva, 2020-11-22:
Hello, I tried to copy .json URL from the esp32s2 branch, but it was exactly the same as from the master branch. I am not very familiar with this system, how do I link the correct .json? Am I supposed to modify it somehow?

lbernstone, 2020-11-22:
https://github.com/espressif/arduino-esp32#installation-instructions

procopio, 2021-01-21:
This works for me perfectly: 
https://www.mischianti.org/2020/12/01/esp32-s2-pinout-specs-and-arduino-ide-configuration-1/