espressif/arduino-esp32 #4208: BLE_client_passkey.ino doesn't work for me

nathan-casabieille, 2020-07-27:
Hello, I am currently trying to retrieve data from a BLE device that has a pin code.

On nRF Connect, I just have to enter the device PIN code and everything works fine.

<p align="center">
[image]
</p>

To get this to work with my esp32, so I tried using [this code](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_passkey/BLE_client_passkey.ino#L21)

So I adapted it to my program, and here is my version of connectToDevice:

```
class MySecurity : public BLESecurityCallbacks {

    uint32_t onPassKeyRequest()
    {
        return (123456);
    }
    void onPassKeyNotify(uint32_t pass_key)
    {
        ESP_LOGE(LOG_TAG, "The passkey Notify number:%d", pass_key);
    }
    bool onConfirmPIN(uint32_t pass_key)
    {
        ESP_LOGI(LOG_TAG, "The passkey YES/NO number:%d", pass_key);
        vTaskDelay(5000);
        return (true);
    }
    bool onSecurityRequest()
    {
        ESP_LOGI(LOG_TAG, "Security Request");
        return (true);
    }
    void onAuthenticationComplete(esp_ble_auth_cmpl_t auth_cmpl)
    {
        if (auth_cmpl.success) {
            ESP_LOGI(LOG_TAG, "remote BD_ADDR:");
            esp_log_buffer_hex(LOG_TAG, auth_cmpl.bd_addr, sizeof(auth_cmpl.bd_addr));
            ESP_LOGI(LOG_TAG, "address type = %d", auth_cmpl.addr_type);
        }
        ESP_LOGI(LOG_TAG, "pair status = %s", auth_cmpl.success ? "success" : "fail");
    }
};

bool connectToDevice(BLEAddress pAddress)
{
    Serial.println("Connecting to device, it may take a few minutes...");

    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new MySecurity());

    BLESecurity* pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);
    pSecurity->setCapability(ESP_IO_CAP_OUT);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

    pClient = BLEDevice::createClient();

    pClient->connect(pAddress);
    if (!pClient->isConnected()) {
        return (false);
    }

    services = pClient->getServices();

    pRemoteService = getService(SERVICE_UUID);
    if (pRemoteService != nullptr) {
        Serial.println("- service [OK]");
    }
    else {
        Serial.println("- service [ERROR]");
        pClient->disconnect();
        return (false);
    }

    characteristics = pRemoteService->getCharacteristics();

    pRemoteDataBufferCharacteristic = getCharacteristic(characteristics, CHAR_DATA_BUFFER_UUID);
    pRemoteRxStateCharacteristic = getCharacteristic(characteristics, CHAR_RX_STATE_UUID);
    if (pRemoteDataBufferCharacteristic != nullptr && pRemoteRxStateCharacteristic != nullptr) {
        Serial.println("- characteristic [OK]");
    }
    else {
        Serial.println("- characteristic [ERROR]");
        pClient->disconnect();
        return (false);
    }
    const uint8_t notificationOn[] = { 0x1, 0x0 };
    pRemoteDataBufferCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
    pRemoteDataBufferCharacteristic->registerForNotify(notifyCallback);
    
    return (true);
}
```
I can connect to the device, but unfortunately, I can't retrieve data (writing in the characteristics does nothing and I receive no notification).

I realized that it never enters the function onPassKeyRequest() so i don't think it connects to device with pin code.

Does anyone have any idea of ​​the mistake I may have made?

Thank You !

chegewara, 2020-07-27:
There may be missing some code in that example, that tells peer device you are able to read pin and use keyboard to send it.

You have few options now:
- post here logs, which may not tell whole story,
- add to your code custom GATTS/GATTC callbacks, you can find more info in other issues about it, and then post logs,
- add code with `ESP_IO_CAP_KBDISP`, more info you can find here:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md

nathan-casabieille, 2020-07-28:
Thank you very much for your help, I will continue on this track !

closed 2020-07-28 by nathan-casabieille

selman-nus, 2021-01-22:
This seems not resolved yet, I am experiencing the same problem. 
Any updates @nathan-casabieille @chegewara  ?

chegewara, 2021-01-25:
As far as i know it is working, just you have to understand how peer device is working and what security is implemented.

selman-nus, 2021-01-26:
I have replied to your comment on [this ](https://github.com/nkolban/esp32-snippets/issues/1045) thread