espressif/arduino-esp32 #4210: How to use PSRAM with tasks

noukee, 2020-07-27:
I use Platformio IDE +arduino-esp32 (1.12.4). And I'm trying to change from esp32-wroom to esp32-wrover to be able to activate BLE, WiFi, TLS and MQTT at the same time.

Hardware:
```
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
```
Use flags:
```
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```
PSRam is enable for ps_malloc()

```
[D][main.cpp:27] logMemory(): Free heap: 332240
[D][main.cpp:28] logMemory(): Used heap: 37008
[D][main.cpp:30] logMemory(): Total PSRAM: 4194252
[D][main.cpp:31] logMemory(): Free PSRAM: 4194252
[D][main.cpp:32] logMemory(): Used PSRAM: 0
```
I have several tasks, Is it possible to allocate tasks on PSRAM with FreeRTOS?

`xTaskCreate(&task, "task", XXX, NULL, 0, &xHandle);`

How could I do? Any recommendation? 

Thank you in advanced

lbernstone, 2020-07-27:
The stack for a task needs to remain in SRAM (and is typically in the 1-8k range).  In arduino-esp32 1.0.4, allocations within the various libraries would need to be manually set to use psram (ps_alloc).  If you use the esp32s2 branch, it has psram mapped into main memory, so large memory allocations will automatically go into the external memory.

noukee, 2020-07-28:
> The stack for a task needs to remain in SRAM (and is typically in the 1-8k range). In arduino-esp32 1.0.4, allocations within the various libraries would need to be manually set to use psram [...]

@Ibernstone, If I have understood correctly the stack for a task uses SRAM but you can allocate manually external memory within the task, correct?

On the other hand, you comment the branchesp32s2, Is it a new chip without BT realased in 2019  or can you use this framework with esp32-wrover-b? Because I need WiFi and BT too.

Thank you!

noukee, 2020-07-29:
Hi, I'm doing tests and with I use malloc It works fine but when I use ps_malloc It gives a memory error. 

```
static StreamDebugger debugger(SerialAT, SerialMon);
static void *psRawMemoryModem = (byte *) ps_malloc(sizeof(TinyGsm));
static TinyGsm *ps_modem = new(psRawMemoryModem) TinyGsm(debugger);

/static TinyGsmClient client(modem);
static void *psRawMemoryClient = (byte *) ps_malloc(sizeof(TinyGsmClient));
static TinyGsmClient *ps_client = new(psRawMemoryClient) TinyGsmClient(*ps_modem);

static void *psRawMemoryMqtt = (byte *) ps_malloc(sizeof(PubSubClient));
static PubSubClient *ps_mqtt = new(psRawMemoryMqtt) PubSubClient(*ps_client);

[....]
void gprs_init()
{
  log_d("[!] Starting gprs task...");
  // Start power management
  if (setupPMU() == false)
  {
    Serial.println("Setting power error");
  }
  // Some start operations
  setupModem();

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);

  xTaskCreate(&gprs_task, "gprs_task", 4000, NULL, 0, &xHandle_gprs);
}

//void gprs_task(void *pvParameters)
void gprs_task()
{
  while (RUNNING) {
    // Restart takes quite some time
    // To skip it, call init() instead of restart()
    SerialMon.println("Initializing modem...");
    ps_modem->restart(); <-- Guru Mediation Error
```

Yo can see the log bellow:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[D][gprs.cpp:188] gprs_init(): [!] Starting gprs task...
Setting power error
Initializing modem...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1ecb  PS      : 0x00060530  A0      : 0x800d1f23  A1      : 0x3ffb1e80  
A2      : 0x00000000  A3      : 0x3f401a39  A4      : 0x00000015  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x8008147e  A9      : 0x3ffb1e60  
A10     : 0x000009ec  A11     : 0x00000000  A12     : 0x000009ec  A13     : 0x00007700  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1ecb:0x3ffb1e80 0x400d1f20:0x3ffb1ea0 0x400d2030:0x3ffb1ed0 0x400d2672:0x3ffb1f00 0x400d2d70:0x3ffb1f50 0x400d12be:0x3ffb1f80 0x400d47ff:0x3ffb1fb0 0x4008b3dd:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57
```

closed 2020-10-17 by stale[bot]

zekageri, 2022-03-08:
Did you solve this memory error?