espressif/arduino-esp32 #4213: Error with LCD with I2C when running with ESP RTOS

ramadhanafif, 2020-07-29:
### Hardware:
Board:							Node 32s
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                         Mac OSX

### Description:
Trying to run a LCD with I2C module attached to it. The examples given by the library creator is working fine, however when I am trying to put a the code inside a (RTOS) task, it will show the text partially, and usually after 3 loops it will show the whole text, then goes back to showing it partially. 

I'm using LCD I2C library by fdebrabander on this [link](https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library) 

To get around this, I usually put it on void setup and void loop. But that's not what I want. 

My initial guess is something goes wrong when I2C-dependent program put inside a task. I cannot pinpoint the problem though. 

The problem is repeatable when I'm running the code below.

### Sketch:
```cpp

#include "Wire.h"
#include "LiquidCrystal_I2C.h"

// Set the LCD address to 0x27 for a 20 chars and 4 line display
LiquidCrystal_I2C lcd(0x27, 20, 4);

void setup() {
  xTaskCreate(
    taskDisplay,    /* Task function. */
    "TaskPause",    /* String with name of task. */
    2048,           /* Stack size in bytes. */
    NULL,           /* Parameter passed as input of the task */
    10,             /* Priority of the task. */
    NULL);          /* Task handle. */

}

void loop() {}

void taskDisplay (void* p) {
  lcd.begin();
  lcd.clear();

  while (1) {
    lcd.setCursor(0, 0);
    lcd.print("some text here");
    vTaskDelay(100);
  }
}
```

ramadhanafif, 2020-08-02:
I found the cause. The I2C is interrupted by void loop, thus you must block the void loop() or use task enter critical (check this [page](https://www.freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html)

closed 2020-08-02 by ramadhanafif

dizcza, 2021-12-05:
I have the same issue thanks @ramadhanafif for pointing this out. But how do you block the void loop()? And could you please show how to use the task enter critical? I'm running into sort of "Core 0 paniced' " problems... I also saw the section about [vTaskSuspendAll](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv415vTaskSuspendAllv) but it didn't help me either.

dizcza, 2021-12-08:
I've come up with a simple trick that does the job:

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void loop() {
    vTaskDelete(NULL);
}
```

`vTaskDelete(NULL)` stops the calling task. The code placed after `vTaskDelete(NULL)` is unreachable. In other words, the `loop` function will be called only once to stop itself. And it won't affect other running threads that have been started in the `setup` function.