espressif/arduino-esp32 #4216: BLE(Server)->Wifi(data transfer)

Dmytro-Bez, 2020-07-30:
I have a problem.
I created a BLE server to collect data from sensors. From the mobile application, I give the command to enable reading data from sensory data. The condition of the task is that when you give the command to start reading data from sensors, data transfer on wifi begins.
When writing code for wifi, the program exceeds the size.
How to solve this problem?

chegewara, 2020-07-30:
Tools->Partition scheme.

Dmytro-Bez, 2020-07-30:
Surprisingly, I do not have such a Partition scheme function in this tab.
What is this feature used for?

lbernstone, 2020-07-30:
Change your board to "ESP32 Dev Module" and you will be able to see what it does.

Dmytro-Bez, 2020-07-31:
Now everything is fine - thanks, but I can't understand how after connection on BLE that the wifi transfer worked?

closed 2020-10-14 by stale[bot]