espressif/arduino-esp32 #4221: [WebServer.h] Get the client's request

nathan-casabieille, 2020-08-03:
Hello, I am currently working on a project which requires to connect multiple esp32 modules via a meshing system.

Each device will therefore create a WIFI access point and connect to the access point of another esp32.

The problem is that sometimes, some devices don't connect to the device they should. This is why I want to host a configuration platform on each esp32 that allows you to modify the SSID of the access point to which the device must connect.

[image]

When the user clicks on the save changes button, here is the request that is formed : `192.168.1.1/submit?dv_ssid=ROOMBOX-56507&ap_ssid=ROOMBOX-1234`

In my code, I define the route like this, so when the client clicks on the button, the esp32 executes the handle_onSubmit function:

server.on("/submit", handle_OnSubmit);

And it is inside this function that I want to parse this request in order to modify the variables accordingly (`String dv_ssid = "ROOMBOX-56507"; String ap_ssid = "ROOMBOX-1234"`).

The problem is that I don't know at all how to get the request (the url). I didn't find anything interesting about it. Does anyone know how I can do this?

Thank You.

sansillusion, 2020-08-03:
Something like :
    String testteu = server.arg("RESET");
    String passme = server.arg("PASSMOI");

nathan-casabieille, 2020-08-04:
@sansillusion Thank you very much, I saw this function in the .h but its name is not really self-explanatory... It works in anyway so it's perfect, thank you again!

closed 2020-08-04 by nathan-casabieille