espressif/arduino-esp32 #4222: The example of 'CaptivePortal' crashed my ESP on connect

jellewie, 2020-08-03:
# Hardware:
Board:							DOIT ESP32 DEVKIT V1, default settings
Core Installation version:			[1.0.4](https://dl.espressif.com/dl/package_esp32_index.json)
IDE name:						Arduino EDE 1.8.9
Flash Frequency:					default settings
PSRAM enabled:                                     default settings
Upload Speed:					921600
Computer OS:                                         Windows 10

#  Description:
Example of CaptivePortal crashed the ESP

### Steps to replicate
1. Uploaded the example CaptivePortal sketch (both the 1.0.4 version and the one on [github](arduino-esp32/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino) do not work)
2. Try to connect with the AP with an (android) device
3. ESP panic'd and rebooted (loop back to step 2 and repeat until you are tired...)

### What I expected
That the example sketch does not crash when a device connects to it.
and that the DNS server will redirect you as it suppose to

# Sketch:
[arduino-esp32/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino)
```#include <WiFi.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 1, 1);
DNSServer dnsServer;
WiFiServer server(80);

String responseHTML = ""
  "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>"
  "<h1>Hello World!</h1><p>This is a captive portal example. All requests will "
  "be redirected here.</p></body></html>";

void setup() { 
  WiFi.disconnect();   //added to start with the wifi off, avoid crashing
  WiFi.mode(WIFI_OFF); //added to start with the wifi off, avoid crashing
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  server.begin();
}

void loop() {
  dnsServer.processNextRequest();
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {
    String currentLine = "";
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        if (c == '\n') {
          if (currentLine.length() == 0) {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();
            client.print(responseHTML);
            break;
          } else {
            currentLine = "";
          }
        } else if (c != '\r') {
          currentLine += c;
        }
      }
    }
    client.stop();
  }
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

# Crash error

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8010feb1  A1      : 0x3ffb3b40  
A2      : 0x3ffcc8a4  A3      : 0x3ffcbfe8  A4      : 0x3ffcb744  A5      : 0x3ffcb724  
A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8010fd54  A9      : 0x3ffb3b00  
A10     : 0x3ffcc8b4  A11     : 0x3ffcbfe8  A12     : 0x3ffb3b4c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3b40 0x4010feae:0x3ffb3b80 0x4011cd35:0x3ffb3ba0 0x40121d3d:0x3ffb3be0 0x40126fda:0x3ffb3c00 0x401108b7:0x3ffb3c20 0x400886e1:0x3ffb3c50`

**=====DECODED**
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x4010feae: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x4011cd35: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
0x40121d3d: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40126fda: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x401108b7: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x400886e1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2020-08-03:
https://github.com/espressif/arduino-esp32/pull/4080

jellewie, 2020-08-03:
If I change the IP to the ESP default IP
`IPAddress apIP(192, 168, **4**, 1);`

and uncomment the softApConfig
`**//**WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));`

**then it does work** on Android and Windows (at least seems to) 

> #4080
About your pull request, here is some feedback (mind I have not tried the code itself)
-SSID can contain spaces
-Furthermore, I also don't really like you removed the 'disconnect' and 'WIFI_OFF' code from it, any reason why?
-Moreover, I find it a stupid idea to change the DNS to the default android DNS, we can be less platform-specific?

jellewie, 2020-10-03:
Ow damm, do I need to write a bot again to keep a ticket open?

jellewie, 2020-12-03:
This issue seems to always occur om some ESP32, But not on others. Have not been able to prove this and such, but it might be some hardware version issue or so?

lvlira, 2021-01-01:
In my case, the sample code in the "DNSServer" lib works in arduino studio. However, when I create a project in VSCode with this code, the error reported by jellewie occurs.

lvlira, 2021-01-09:
> In my case, the CaptivePortal code in the "DNSServer" lib works in Arduino IDE 1.8.13. However, when I create a project in VSCode with this code, the error reported by jellewie occurs.
> 
> Task: Trying run the CaptivePortal code using VSCode
> 
> System:
> VSCode version: 1.52.1
> Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
> Date: 2020-12-16T16:32:10.090Z
> Electron: 9.3.5
> Chrome: 83.0.4103.122
> Node.js: 12.14.1
> V8: 8.3.110.13-electron.0
> OS: Linux x64 5.4.0-58-generic
> PlatformIO: 1.319.687
> C++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
> 
> Board: esp-wrover-kit
> 
> Problem (only VSCode with PlatformIO) : Guru Meditation Error After uploading the code to ESP32 and trying to connect to the wifi network using a Android 8.0 smartphone
> 
> Main.cpp:
> #include <WiFi.h>
> #include <DNSServer.h>
> 
> const byte DNS_PORT = 53;
> IPAddress apIP(8,8,4,4); // The default android DNS
> DNSServer dnsServer;
> WiFiServer server(80);
> 
> String responseHTML = ""
> "<title>CaptivePortal</title>"
> "
> # Hello World!
> 
> This is a captive portal example. All requests will "
> "be redirected here.
> ";
> 
> void setup() {
> 
> WiFi.mode(WIFI_AP);
> WiFi.softAP("ESP32-DNSServer");
> WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
> 
> // if DNSServer is started with "_" for domain name, it will reply with // provided IP to all DNS request dnsServer.start(DNS_PORT, "_", apIP);
> 
> server.begin();
> }
> 
> void loop() {
> dnsServer.processNextRequest();
> WiFiClient client = server.available(); // listen for incoming clients
> 
> if (client) {
> String currentLine = "";
> while (client.connected()) {
> if (client.available()) {
> char c = client.read();
> if (c == '\n') {
> 
> ```
>       if (currentLine.length() == 0) {
>         client.println("HTTP/1.1 200 OK");
>         client.println("Content-type:text/html");
>         client.println();
>         client.print(responseHTML);
>         break;
>       } else {
>         currentLine = "";
>       }
>     } else if (c != '\r') {
>       currentLine += c;
>     }
>   }
> }
> client.stop();
> ```
> 
> }
> }
> 
> PLATFORMIO.INI:
> [env:esp32dev]
> platform = espressif32
> board = esp32dev
> framework = arduino
> monitor_speed = 115200
> 
> ERROR MESSAGE:
> ets Jun 8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:8896
> load:0x40080400,len:5828
> entry 0x400806ac
> dhcps: send_offer>>udp_sendto result 0
> Guru Meditation Error: Core 0 panic'ed (InstrFetchProhibited). Exception was unhandled.
> Core 0 register dump:
> PC : 0x00000000 PS : 0x00060e30 A0 : 0x80110781 A1 : 0x3ffb3b40
> A2 : 0x3ffcbe3c A3 : 0x3ffcc070 A4 : 0x3ffcc9c8 A5 : 0x3ffcc9a8
> A6 : 0x05040808 A7 : 0x0f040808 A8 : 0x80110624 A9 : 0x3ffb3b00
> A10 : 0x3ffcbe4c A11 : 0x3ffcc070 A12 : 0x3ffb3b4c A13 : 0x00000044
> A14 : 0x00000001 A15 : 0x00000006 SAR : 0x00000010 EXCCAUSE: 0x00000014
> EXCVADDR: 0x00000000 LBEG : 0x4000c349 LEND : 0x4000c36b LCOUNT : 0x00000000
> 
> Backtrace: 0x00000000:0x3ffb3b40 0x4011077e:0x3ffb3b80 0x4011d605:0x3ffb3ba0 0x4012260d:0x3ffb3be0 0x401278aa:0x3ffb3c00 0x40111187:0x3ffb3c20 0x400886e1:0x3ffb3c50
> 
> Rebooting...
> ets Jun 8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:8896
> load:0x40080400,len:5828
> entry 0x400806ac

Solved!

I put a 2s delay to allow the softAP command to stabilize.
  WiFi.softAP("ESP32-DNSServer");
  delay(2000);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

zekageri, 2022-09-02:
Still happens

VojtechBartoska, 2022-09-02:
@zekageri testes on 2.0.4?

zekageri, 2022-09-02:
> @zekageri tested on 2.0.4?

```
Processing esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)
------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (5.1.1) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, 
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.220825 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

framework is 2.0.4 yes.

vortigont, 2022-12-11:
You may try updated Captive Portal example from [DNSServerAsync](https://github.com/vortigont/ESP32-DNSServerAsync) lib.
Could also be easily adopted to bundled DNSServer lib.

zekageri, 2022-12-11:
I will give it a try. Hope it works with Async we server by me_no_dev

zekageri, 2022-12-11:
> You may try updated Captive Portal example from [DNSServerAsync](https://github.com/vortigont/ESP32-DNSServerAsync) lib. Could also be easily adopted to bundled DNSServer lib.

First try

```cpp
assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

Backtrace:0x40083cbd:0x3ffb55c00x4008ece5:0x3ffb55e0 0x40094509:0x3ffb5600 0x4011ba5e:0x3ffb5730 0x4011bb0c:0x3ffb5750 0x40186042:0x3ffb5770 0x40118398:0x3ffb5790 

  #0  0x40083cbd:0x3ffb55c0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402      
  #1  0x4008ece5:0x3ffb55e0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x40094509:0x3ffb5600 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85        
  #3  0x4011ba5e:0x3ffb5730 in tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:951
      (inlined by) tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:931 
  #4  0x4011bb0c:0x3ffb5750 in tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:991 
  #5  0x40186042:0x3ffb5770 in _tcp_recved_api(tcpip_api_call_data*) at lib/AsyncTCP/src/AsyncTCP.cpp:431
  #6  0x40118398:0x3ffb5790 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:172
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 0000000000000000

Rebooting...
```

vortigont, 2022-12-11:
@zekageri are you sure you've build and example from [DNSServerAsync](https://github.com/vortigont/ESP32-DNSServerAsync/tree/main/examples/CaptivePortal)?

In your trace there is an AsyncTCP call. That example does not use AsyncTCP lib.
```
#5  0x40186042:0x3ffb5770 in _tcp_recved_api(tcpip_api_call_data*) at lib/AsyncTCP/src/AsyncTCP.cpp:431
      (inlined by) tcpi
```

zekageri, 2022-12-11:
I had only this handler:

```cpp
captiveSerer.on("/", HTTP_GET, [this](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response =  request->beginResponse(
        LittleFS,
        WELCOME_PAGE_PATH,
        sGlobal.getMimeType(WELCOME_PAGE_PATH)
    );
    request->send(response);
});
```

I added `/portal` endpoint as in the example, it seems much more stable now. 
I have two endpoint now:

```cpp
captiveSerer.on("/", HTTP_GET, [this](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response =  request->beginResponse(
        LittleFS,
        WELCOME_PAGE_PATH,
        sGlobal.getMimeType(WELCOME_PAGE_PATH)
    );
    request->send(response);
});

captiveSerer.on("/portal", HTTP_GET, [this](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response =  request->beginResponse(
        LittleFS,
        WELCOME_PAGE_PATH,
        sGlobal.getMimeType(WELCOME_PAGE_PATH)
    );
    request->send(response);
});
```

**Will test it further.**

zekageri, 2022-12-11:
> @zekageri are you sure you've build and example from [DNSServerAsync](https://github.com/vortigont/ESP32-DNSServerAsync/tree/main/examples/CaptivePortal)?
>
> In your trace there is an AsyncTCP call. [...]

But i use it

zekageri, 2022-12-11:
So far the captive portal itself is working on android and windows. It pops up the webpage with Async Webserver too. And the crashes are much fewer than before.

vortigont, 2022-12-11:
> But i use it

Looks like you have lot's of other things like littlefs and not sure what is `sGlobal.getMimeType(WELCOME_PAGE_PATH)`. There could be lot's of places to hit a dangling pointer when using AsyncWebserver.
My humble advise is to replace page content and type with static "char-something" and test this way to mitigate any dependent code.
Captive portal reqs should be really really fast served due to modern devices generates tons of requests for new wifi connection.

zekageri, 2022-12-11:
Yes, i use LittleFS to serve webpages from the file system.

`sGlobal.getMimeType(WELCOME_PAGE_PATH)` will determine a correct mime type for a given file, nothing much just a couple of if statements.

I use captive portal for first configuration. The user will post a config object with HTTP_POST.

vortigont, 2022-12-11:
@zekageri I understand, but here discussing an issue with "CaptivePortal example" not "I have a project with async server, littlefs + something" and it fails. Pls, do not consider me being rude, just trying to narrow down the issue. Suspect that the root cause for an old example is that `WiFiServer` can't handle multiple concurrent tcp sessions which happens with captive portal when lot's of req's are redirected by DNSServer. An updated example could stand a better chance (or not).

> `sGlobal.getMimeType(WELCOME_PAGE_PATH)` will determine a correct mime type for a given file, nothing much just a couple of if statements.

so suppose it would not be a problem to stub it with `const char*` strings and test if it helps here or not, right? 
Isolate the problem and reduce the complexity is a way to go when troubleshooting. Otherwise you maybe looking for an issue in a wrong place. Looking into stack trace a crash is somewhere on attempt to release a heap resource. Either mem region is corrupted or reference is wrong for some dynamically allocated object.
Cheers!

zekageri, 2022-12-11:
@vortigont Thank you for the positive feedback. I appreciate.

I understand that, but this issue only happens on captive portal and nowhere else in my code. I have literally hundreads of endpoints using this exact same model with Async webserver and littlefs and getMimeType and such things. I have an admin page where the page has a concurrent request more than 15. I know i have to narrow down the problem, but this is the narrowed down part. I suspect that the problem will be somewhere in lwip because most of the error's content are related to it in some way.

If my code runs without captive portal it works in every imaginable mode. I'm also using ethernet and wifi and AP mode. In every case it is stable, except at captive portal connection. And this snippet is nowhere near the complexity of my any other frequently used endpoints.

PilnyTomas, 2023-01-19:
I tried the example on the latest version of Arduino-esp32 with ESP32 and an android phone and it works for me.
Could you please retest it on the current version?
If it does not work, please tell us more about your setup.

closed 2023-02-14 by PilnyTomas

zekageri, 2023-02-14:
Looks good to me so far.