espressif/arduino-esp32 #4226: When using AP mode, serial port is being used to issue unwanted message

brownrb, 2020-08-04:
When configuring the ESP32 as an Access Point, a library routine issues the following message on the Serial port when a TCPIP client connects to the AP on the ESP32

dhcps: send_offer>>udp_sendto result 0

Can this be somehow turned off. Why is it enabled by default?

I use a combination of the AP and serial port to manage devices. Having a library issue an unwanted serial port message is not something that I should have to deal with in my code IMHO.

Is there some way to disable this message?

I am using Arduino 1.8.13 and the latest Arduino Core for ESP32

Regards
Robert

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

jellewie, 2020-08-04:
Can confirm, sometimes also
`dhcps: send_offer>>udp_sendto result 0`

Maybe related to #2285 Invalid Event when Client connect AP + debug output 
The only kinda reference I found so far is 
https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L527
Which states
`#if DHCPS_DEBUG
        DHCPS_LOG("dhcps: send_offer>>pbuf_alloc failed\n");
#endif`
so you could try to undefine DHCPS_DEBUG as a workaround.

Also as a side note, you should properly fill in the issue.

atanisoft, 2020-08-04:
same as https://github.com/espressif/arduino-esp32/issues/4027 and https://github.com/espressif/esp-idf/issues/2924. This fix was not backported to the release of esp-idf that arduino-esp32 is using in v1.0.4 but the fix should be available on the esp32s2 branch of arduino-esp32.

scubachristopher, 2020-10-18:
When will this fix go out in a general release?

atanisoft, 2020-10-18:
@scubachristopher not until esp32s2 branch is merged to master.

scubachristopher, 2020-11-08:
@atanisoft -- Any idea what date that will happen?

atanisoft, 2020-11-08:
Not sure

lbernstone, 2020-11-08:
I think the fix is in 1.0.5.

scubachristopher, 2020-12-04:
Ok, so I'm on PlatformIO 5.0.3.  How do I bring in this fix?

jellewie, 2021-02-04:
I don't know how it works in PIO I just use Arduino IDE

brownrb, 2021-06-21:
Bug is still there. Still waiting for an updated core for esp32
It is coming up to 10 months now.

closed 2021-09-19 by stale[bot]