espressif/arduino-esp32 #4227: I2c i2c_isr_handler_default cause watchdog timeout

DamithVG, 2020-08-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					       115200 
Computer OS:                                         Windows 10

### Description:
when running a code, sometimes i2c_isr_handler_default() function executes over and over again and causes firing the interrupt watchdog. This doesn't happen periodically. Sometimes it is 2 3 hours after ESP starts and sometimes it is just 20 minutes. 
According to the debug text, activeInt variable of  esp32-hal-i2c.c file is  4. 

When i examine the esp32-hal-i2c.c file, in line 909, i noticed that interrupts were handled in an infinite while loop. It only exits when activeInt is 0. 
Now look the if condition in line 1003.  

if(activeInt) { // clear unhandled if possible?  What about Disabling interrupt?
            p_i2c->dev->int_clr.val = activeInt;
            log_e("unknown int=%x",activeInt);
            // disable unhandled IRQ,
            p_i2c->dev->int_ena.val = p_i2c->dev->int_ena.val & (~activeInt);
        }

If this condition is the only condition becomes true, the while loop never exits because activeInt will not set to 0 and ISR executes until INT WDT fires. 

This is the case as i understand. Please correct me if im wrong. 

Thank you.

### Debug Messages:
```
[03/08/20 - 16:03:09:283] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:294] 

[03/08/20 - 16:03:09:294] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:304] 

[03/08/20 - 16:03:09:304] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:304] 

[03/08/20 - 16:03:09:304] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:304] 

[03/08/20 - 16:03:09:304] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:314] 

[03/08/20 - 16:03:09:314] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:323] 

[03/08/20 - 16:03:09:323] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:323] 

[03/08/20 - 16:03:09:323] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:333] 

[03/08/20 - 16:03:09:333] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:333] 

[03/08/20 - 16:03:09:333] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:344] 

[03/08/20 - 16:03:09:344] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:353] 

[03/08/20 - 16:03:09:353] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:353] 

[03/08/20 - 16:03:09:353] [E][esp32-hal-i2c.c:1005] i2c_isr_handler_default(): unknown int=4
[03/08/20 - 16:03:09:363] 

[... 176 lines cut ...]
[03/08/20 - 16:03:09:744] 

[03/08/20 - 16:03:09:744] 
[03/08/20 - 16:03:09:753] 

[03/08/20 - 16:03:09:753] Rebooting...
[03/08/20 - 16:03:09:753] 
```

decoded backtrace 

Decoding stack results
0x400e216e: uartBegin at C:\Users\VEGA_1010\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 214
0x40081a7a: pinMatrixInDetach at C:\Users\VEGA_1010\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-matrix.c line 40
0x40082b11: esp_intr_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c line 801
0x4008a669: prvAddNewTaskToReadyList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 1149
0x4008c3b3: multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 155
0x4008cb64: multi_heap_get_allocated_size at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 295
0x40085368: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x4008539d: heap_caps_realloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 282
0x40085861: lock_init_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 85
0x400e212d: uartBegin at C:\Users\VEGA_1010\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 207
0x400dbfde: ping_start(IPAddress, int, int, int, int, ping_option*) at C:\Users\VEGA_1010\Documents\Arduino\libraries\ESP32Ping-master\ping.cpp line 121
0x400dc33d: WiFiClient::connect(char const*, unsigned short, int) at C:\Users\VEGA_1010\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/IPAddress.h line 29
0x400894d1: xPortSysTickHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 284

atanisoft, 2020-08-04:
You can't call log_e inside an isr. The backtrace points to the uart failing to transmit and not i2c at fault.

However, in the snippet you posted from i2c, the first line inside that if block will clear the interrupt and the line after the log will disable it from being raised again.

The activeInt variable should be reset by assignment inside the loop somewhere.

DamithVG, 2020-08-04:
> 
> 
> You can't call log_e inside an isr. The backtrace points to the uart failing to transmit and not i2c at fault.
>
But int the I2c Hal.c file, theydid it.  https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c 
And also please forgive me, i decoded part of the backtrace, now i edited it with full decode result. Semms like it something related to heap. 
Default Int wdt timeout is 300ms. As you can see, in the log file timestamps,    Watchdog fires exactly 300ms after the i2c_isr_handler_default() execution started, 

> However, in the snippet you posted from i2c, the first line inside that if block will clear the interrupt and the line after the log will
 disable it from being raised again.
>

Yes that's true, But ISR never exits. 

> The activeInt variable should be reset by assignment inside the loop somewhere.
>
Yes that should, But couldn't find a resetting line int hal i2c.c file.

atanisoft, 2020-08-04:
> But int the I2c Hal.c file, theydid it

Yes, I see that. It should be isr_log_e() instead.

> And also please forgive me, i decoded part of the backtrace, now i edited it with full decode result. Seems like something related to heap.

The updated decode doesn't appear related to I2C but instead due to re-initialization of the UART which looks like an issue that was reported already (though I can't find it currently).

> Yes that should, But couldn't find a resetting line in hal i2c.c file.

inside the if statement where the log_e call is add an assignment to set activeInt to zero. This block shouldn't ever be entered under normal circumstances though.

DamithVG, 2020-08-04:
> The updated decode doesn't appear related to I2C but instead due to re-initialization of the UART which looks like an issue that was reported already (though I can't find it currently).
> 
If you find the issue, please post it here

> inside the if statement where the log_e call is add an assignment to set activeInt to zero. This block shouldn't ever be entered under normal circumstances though.
>
It is a Rx FIFO overflow. Dont know what cause it. I think wire library should take care of overflows. However for some reason, if it happens INT WDT timeout is inevitable. 

Thank you

DamithVG, 2020-08-04:
> Yes, I see that. It should be isr_log_e() instead.
> 
Is it possible that  log_e cause the UART malfunction?

atanisoft, 2020-08-04:
> Is it possible that log_e cause the UART malfunction?

Being called inside an ISR context very likely it is not helping.

> It is a Rx FIFO overflow.

that should be handled elsewhere in the ISR code I believe.

> Dont know what cause it. I think wire library should take care of overflows.

Often it is a misbehaving library that uses Wire, ie it doesn't check error codes etc...

lbernstone, 2020-08-04:
Fix Serial.end: https://github.com/espressif/arduino-esp32/pull/3894

closed 2020-08-11 by DamithVG