espressif/arduino-esp32 #4228: Association sleep pattern?

matthewlai, 2020-08-04:
### Hardware:
Board:							Custom board with ESP32-WROOM-32. Schematics: https://drive.google.com/file/d/1B1t2fKCH75Z6QrWmrEZD0VwUjEYmn7Ae/view?usp=sharing
Core Installation/update date:			Aug 4, 2020
IDE name:							Arduino IDE
Flash Frequency:					80 MHz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
I am trying to reduce power consumption while maintaining WiFi association. In previous revisions of the datasheet there is mentioning of an "associative sleep pattern" that allows light sleep between DTIM interval. Is that supported with arduino-esp32?

I am getting 60mA with the sketch below. Is there a way to lower it while maintaining association?

Thanks!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti WiFiMulti;

const char* ssid = "x";
const char* pass = "x";

void setup()
{  
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network
  WiFiMulti.addAP(ssid, pass);

  Serial.println();
  Serial.println();
  Serial.print("Waiting for WiFi... ");

  while (WiFiMulti.run() != WL_CONNECTED) {
      Serial.print(".");
      delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  delay(1000);
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: xxx

Waiting for WiFi... [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 10 networks found
[...]
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xxx SSID: xxx Channal: 6 (-42)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: xxx, MASK: 255.255.255.0, GW: xxx
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: xxx
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: xxx
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: xxx
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 6
```

lbernstone, 2020-08-04:
This forum is for issues with arduino code.  For general esp32 questions, please go to https://esp32.com or read the esp-idf documentation on [sleep](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-reference/system/sleep_modes.html) and [pm](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-reference/system/power_management.html).

matthewlai, 2020-08-04:
@lbernstone I believe this is arduino-esp32-specific. It's about how/whether those features are accessible through Arduino (many of them aren't because they interact with peripheral clocks).

manuprendlair, 2020-08-12:
@matthewlai As I know power management is not enable in Arduino core.
You must use an esp idf project with Arduino as component and use menuconfig to enable power management

closed 2020-10-28 by stale[bot]