espressif/arduino-esp32 #4229: Can't connect to eduroam WPA2 Enterprise network

amarotica, 2020-08-05:
Make your question, not a Statement, inclusive. Include all pertinent information:

I am trying to connect an ESP-WROOM-32 devkit to an eduroam enterprise wifi network

I am receiving the error: 
E (349) wpa: Method private structure allocated failure
....................................................

The code I am trying:
`// eduroam WPA2 Enterprise, TTLS, no certificate, MSCHAPv2(no EAP)
#include "esp_wpa2.h"
#include <WiFi.h>
#define EAP_IDENTITY "xxxxx" // ID eg: "jonny" not jonny@univ.xx
#define EAP_PASSWORD "xxxxx"
#define MENOPC "ESP32 Board"

String line;
const char* ssid = "eduroam";
WiFiClient client;

void setup() {
Serial.begin(115200);
delay(10);
Serial.println();
Serial.println(ssid);
WiFi.disconnect(true);
WiFi.mode(WIFI_STA);
esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY));
esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY));
esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));
esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
esp_wifi_sta_wpa2_ent_enable(&config);

Serial.println("MAC address: ");
Serial.println(WiFi.macAddress());
WiFi.begin(ssid);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());

}

void loop() {
while (client.available()) {
char c = client.read();
Serial.write(c);
}

if (!client.connected()) {
if (client.connect("my-site", 3000)) {
client.println("GET /sensores?temperatura=123&id=ESP32 HTTP/1.1");
client.println("Connection: close");
client.println();
delay(2000);
}
}
}`
Error message:
`ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

eduroam
MAC address: 
24:6F:28:A9:71:FC
E (349) wpa: Method private structure allocated failure

....................................`

### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					240Mhz
PSRAM enabled:                                     no
Computer OS:                                        Windows 10

### Description:
Describe your problem here

### Debug Messages:
```
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

eduroam
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
MAC address: 
24:6F:28:A9:71:FC
E (495) wpa: Method private structure allocated failure

.....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 23 - 802_1X_AUTH_FAILED
...............```

Any help would be greatly appreciated. I've tried every example sketch I can find...

lbernstone, 2020-08-06:
That error Method private structure allocated failure seems to be the source of the problem.  Change the enable line to:
```
esp_err_t result = esp_wifi_sta_wpa2_ent_enable(&config);
Serial.println(esp_err_to_name(result));
```

amarotica, 2020-08-07:
> That error Method private structure allocated failure seems to be the source of the problem. Change the enable line to:
> 
> ```
> esp_err_t result = esp_wifi_sta_wpa2_ent_enable(&config);
> Serial.println(esp_err_to_name(result));
> ```

Thanks for the reply. I tried it and this is the output:

eduroam
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
ESP_OK
MAC address: 
24:6F:28:A9:71:FC
E (357) wpa: Method private structure allocated failure

..[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 23 - 802_1X_AUTH_FAILED
.........................

Didn't seem to change anything, including an error message

lbernstone, 2020-08-07:
That error is coming from inside the wpa blob.  You will have to open an issue at https://github.com/espressif/esp-idf/issues for Espressif to look at the code.

amarotica, 2020-08-07:
@lbernstone alright, thanks for the info, I will add it on esp-idf/issues. Hopefully they'll be able to make something of it since it's been an issue for many people over the last few years and hopefully we could be some of the last...

closed 2020-08-07 by amarotica