espressif/arduino-esp32 #4231: Why Code is crashing due to external interrupt while writing to internal EEPROM. Why addition of IRAM_ATTR also not resolving this issue.

KreakEmp, 2020-08-06:
Code is crashing due to external interrupt while writing to internal EEPROM. I have gone through the issues and found that addition of IRAM_ATTR will move the isr to RAM memory and it will work fine but even if I am adding that it's not working.

Here I am adding my ISR routine. Kindly let me know what's going wrong and how it can corrected.

I am trying to read rain tip count from davis rainfall sensor using external interrupt and also frequently updating the internal eeprom with error code. 

Using DOIT ESP32 DEVKIT V1 in arduino platform.

Code is crashing when interrupt and writing to the internal EEPROM is happening same time. I have gone through this with different forum and got the solution that IRAM_ATTR attribute needs to be added before function code. But after adding this to the ISR, still there is no improvement code is crashing randomly.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz?
                                   
Upload Speed:					921600
Computer OS:                                        Windows 10

### Sketch: 

#include <Arduino.h>
#include <EEPROM.h:>

#define DAVIS_RAIN_GAUGE_PIN   18
#define ISR_DEBOUNC_TIMEOUT   20                             //in millis
volatile unsigned long currentMillis;
volatile unsigned long contactTime;
volatile unsigned long tipCount;

static void IRAM_ATTR isr_count_change() {
	currentMillis = millis();
	if( currentMillis - contactTime > ISR_DEBOUNC_TIMEOUT){
	        tipCount++;
				
	}	
}

void setup() {
       Serial.begin(115200);
       Serial.print(F("Code Started. "));   
       attachInterrupt(digitalPinToInterrupt(DAVIS_RAIN_GAUGE_PIN), isr_count_change, CHANGE);
      EEPROM.begin(4096);
}

void loop() {
        EEPROM.put(100, tipCount);
        Serial.print(F("Rain tip count : "));            Serial.println(int(tipCount));
       delay(2000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40135f74: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40135f74: bad00bad bad00bad bad00bad

DougArmstrong, 2020-08-06:
I couple of things to check that might be related, this example code addresses a couple of them. They this as a base and see if it helps. Also be sure if you are using internal pull-ups you are on pins that support them.

`const byte interruptPin = 25;
volatile int interruptCounter = 0;
int numberOfInterrupts = 0; 
portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED; 

void IRAM_ATTR handleInterrupt() 
{  
portENTER_CRITICAL_ISR(&mux);  
interruptCounter++;  
portEXIT_CRITICAL_ISR(&mux);
} 

void setup() {   
Serial.begin(115200);  
Serial.println("Monitoring interrupts: ");  
pinMode(interruptPin, INPUT_PULLUP);  
attachInterrupt(digitalPinToInterrupt(interruptPin), handleInterrupt, FALLING); 
} 

void loop() {   
if(interruptCounter>0){       
portENTER_CRITICAL(&mux);      
interruptCounter--;      
portEXIT_CRITICAL(&mux);       
numberOfInterrupts++;      
Serial.print("An interrupt has occurred. Total: ");      
Serial.println(numberOfInterrupts);  
}

}` 
--

KreakEmp, 2020-08-07:
Thanks DougArmstrong for your replay. Can you please explain a bit what you are trying to implement and what possible reason it is not working. Meanwhile, I will test this sample and let you know.

DougArmstrong, 2020-08-07:
The code I offered is a common sample, found here: https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/  It contains a great discussion of the issues. 

A couple of comments that might be worth further research:

var++/-- operations are not necessary atomic which means that these values might be corrupted or in other words interrupted.  Using semaphore locks, will maintain the integrity of these values.  

Unlike regular Arduino systems, ESP32 uses FreeRTOS to manage many of its on board systems behind the scenes, so a bit more coding discipline is sometimes required.  While I am not sure why your code is crashing, tightening up the structure would be my first suggestion.

Couple of comments: It appears you are interfacing to a Davis teeter-toter rain gauge.  Using an ISR is a clever way to go, but of course the rain gauge is very slow.  This could be polled in a task or timer tick ISR easily enough.  I see you are filtering out switch bounce which is always a good call.  If you are measuring wind speed ISR response time may be more critical depending on if you do pulse counting or Pulse Width measurement.  

Finally there are some interesting discussions on which pins function correctly as interrupt inputs.  A few in the mid 30 are not always stable.

Finally, check out including FreeRTOS as a way of structuring your programs to use tasks, queues and timers pays off.  This works great in the Arduino IDE as well as vMicro inside visual studio.

I hope some of this helps.

atanisoft, 2020-08-07:
@KreakEmp The reason your code is crashing is due to the pin change interrupt firing while a flash operation is in progress. The EEPROM library is a very thin wrapper around a virtual EEPROM block stored in flash. You can work around this in a few ways, one option being a mutex that @DougArmstrong referenced. Another being remove the IRAM_ATTR tag from the pin change interrupt. There are drawbacks to both of these options.

Based on your usage though you might be better off using the [PCNT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/pcnt.html) to count the pulses on the IO pin which will not require you to use an ISR at all. Even though this is an ESP-IDF function it is available in the arduino environment.

closed 2020-10-22 by stale[bot]