espressif/arduino-esp32 #4233: ESP32 SOLO 1 problems with Arduino IDE

aledoors, 2020-08-06:
Hi all
I carry out the step-by-step procedure to be able to use my ESP32 SOLO 1 on Arduino IDE, but after compiling and uploading my program to the microcontroller I have the error that my ES32 has a single core and I don't know where I can change its configuration to not have this problem. Using ESP IDF I execute the instruction "idf.py menuconfig" and I can select to use one core, but on Arduino IDE I can't do it.
Thanks in advance.

bfaliszek, 2020-08-07:
try: https://github.com/lbernstone/arduino-esp32-solo

closed 2020-10-22 by stale[bot]

bfaliszek, 2021-02-26:
The problem returned with the Arduino core for the ESP32 - 1.0.5

sixtyfive, 2021-11-15:
It seems #5183 is the official place for this now.