espressif/arduino-esp32 #4243: BLE crashes when IDF functions used while using Arduino as a component

Sylphe88, 2020-08-11:
Hi there,

I've been struggling for a few hours to understand why the BLE functions used in my code would not work on my ESP32.
Using:
IDF 4.0.1
Arduino as a component (1.0.4 - last release) 

Basically, the few lines of code that were crashing :
`ESP_ERROR_CHECK(nvs_flash_init_partition("nvs"));
 ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
 esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();

 ESP_ERROR_CHECK(esp_bt_controller_init(&bt_cfg));`

Crash trace is attached: [PRC_20200811-112531.log](https://github.com/espressif/arduino-esp32/files/5056270/PRC_20200811-112531.log)

part of sdkconfig worth mentioning:
`
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
`

These lines of code work fine in a pure ESP-IDF environment, so I figured there was something going on with the Arduino cores
After quite a while, i ended up spotting the issue here:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L222

Basically at the time InitArduino() is called, if the Bluetooth is unused all the memory is released (for CLASSIC_BT and BLE), which obviously prevents subsequent calls to the BLE stack from working. Commenting the mem_release function in the esp32-hal-misc.c removes the crash and the BLE stack then works as expected.

Does anyone know why these lines in esp32-hal-misc.c were added (except for memory optimizations)? IMHO we should change this behavior (or document it) so people using Arduino as a component are not misleaded.

Best,

chegewara, 2020-08-12:
Lets start from beginning:
>IDF 4.0.1
Arduino as a component (1.0.4 - last release)

Those versions are not compatible. v1.0.4 is based on esp-idf v3.2 or v3.3, but that is not a big problem.

Now, maybe try to add `BLEDevice::init()` in setup, somewhere at beginning.

manuprendlair, 2020-08-12:
@chegewara is right. the correct ESP iDF version is the commit number 367c3c09c.
But I am not sure the problem come from the ESP IDF version.
Last years I try to use BLE example with arduino as component in ESP IDF and it not working but at this time I was not able to find why.
@Sylphe88 could you try with the right esp idf versioni and confirm it crash also (pretty sure it is!). If your solution solve the problem this a good news 'cause I will try to play again with ble.
@me-no-dev is I think the only who could answer to the @Sylphe88 question!
@me-no-dev  could you give us some informations about ?

Sylphe88, 2020-08-13:
> 
> 
> Lets start from beginning:
> 
> > IDF 4.0.1
> > Arduino as a component (1.0.4 - last release)
> 
> Those versions are not compatible. v1.0.4 is based on esp-idf v3.2 or v3.3, but that is not a big problem.
> 
> Now, maybe try to add `BLEDevice::init()` in setup, somewhere at beginning.

Sorry, filled in the versions too quickly, I'm using the idf-release/v4.0 branch of the Arduino repo, I'll edit my OP.

About your workaround, it would totally work, but I typically do not want to load the Arduino BLE classes as they use far too much RAM for my app compared to the IDF stack (the reason why I'm using the Arduino cores is essentially for the Serial and the SPIFFS classes).

@manuprendlair I did manage to make it work, I'm only bringing up this thread to figure out a decent solution that would enable a IDF project using Arduino cores to use the esp_ble_xxx functions without "invalid state" errors ;)

closed 2020-10-28 by stale[bot]