espressif/arduino-esp32 #4245: Arduino as IDF component is causing fails when entering the menuconfig in Windows:  " Failed to resolve component 'ethernet' "

athusrv, 2020-08-11:
By following this guide [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md), I got errors when trying to run the menuconfig command [`idf.py menuconfig`].

The error message is: `Failed to resolve component 'ethernet'`

When I don't use the arduino as a IDF component (pure IDF project), everything works great.

Does anyone know the solution for this problem? Here is the complete log output that I got in the terminal:

```
$ idf.py menuconfig
Python requirements from D:\Athus\workspace\esp-idf\requirements.txt are satisfied.
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Error at D:/Athus/workspace/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.
Call Stack (most recent call first):
  D:/Athus/workspace/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  D:/Athus/workspace/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  D:/Athus/workspace/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "D:/Athus/workspace/asimov/code/hello_world/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja
D:\Athus\.espressif\tools\cmake\3.13.4\bin\cmake.exe -SD:\Athus\workspace\asimov\code\hello_world -BD:\Athus\workspace\asimov\code\hello_world\build
ninja: error: rebuilding 'build.ninja': subcommand failed
Checking Python dependencies...
Executing action: menuconfig
Running ninja in directory d:\athus\workspace\asimov\code\hello_world\build
Executing "ninja menuconfig"...
ninja failed with exit code 1
```

Environment:
- Windows 7
- cmd.exe, powershell, git bash
- esp-idf v4.0

lbernstone, 2020-08-11:
Arduino as a component can only be guaranteed to compile on release versions of arduino-esp32.  That would currently be IDF 3.3 (IDF_COMMIT=4e27cbb89).  It will also work with IDF 4.2 with the esp32s2 branch, but that is beta code.  In between versions are a moving target, so you would be required to resolve the changes.  The ethernet component was renamed to esp_eth in 4.0.  Once you fix that issue, you will find 100 more.

athusrv, 2020-08-12:
Thank you @lbernstone . From your answer, I was able to make it work.
Here is what I did:
- Make sure I had the ESP-IDF from `release/v4.2` branch (along with Git and Python, obviously) 
- Ran the `install` script from `esp-idf` folder (for Windows it's `install.bat`)
- I used the `hello_world` example from `esp-idf\examples\get-started`. So I made a copy of the `hello_world` folder to somewhere
- Inside the `hello_world` folder I created another folder called `components`
- Inside this `components` folder I cloned the arduino-esp32 repo (`git clone https://github.com/espressif/arduino-esp32.git arduino`)
- Inside the `arduino` folder I did those steps:
  - `git fetch --all`
  - `git checkout esp32s2`
  - `git submodule update --init --recursive`
- I went back to the `hello_world` folder and ran `idf.py menuconfig` (but first I ran the `export.bat` script from the `esp-idf` folder)
- I could see the `Arduino Configuration` option there. Inside that option I enabled `Autostart Arduino setup and loop on boot`
- I also enabled the `Include only specific Arduino libraries` and checked only those I needed
- Also, in the root menu, I went to `Component Config -> mbedTLS -> TLS Key Exchange Methods` and checked the option `Enable pre-shared-key ciphersuites` and then `Enable PSK based ciphersuite modes`
- I saved the configuration and closed the `menuconfig`
- Then, I opened the `main` folder that inside `hello_world` folder and renamed the `hello_world_main.c` to `main.cpp`
  - Also, I changed the `CMakeLists.txt` inside the `main` folder. In this file, I also renamed the part that has `hello_world_main.c` to `main.cpp`
- I pasted the following code to the `main.cpp` file:
```
#include "Arduino.h"
#include "sdkconfig.h"

void setup() {
    Serial.begin(115200);
}

void loop() {
    Serial.println("Hello");
    delay(1000);
}
```

- I ran `idf.py build` and the project was built successfully
- Then I ran `idf.py flash monitor` and I could see the `Hello` messages in the terminal :partying_face:

lbernstone, 2020-08-12:
Your process is pretty much exactly what [lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) does for you.

athusrv, 2020-08-12:
@lbernstone isn't lib-builder aimed to be used on UNIX based OS? I'm using Windows.

The ESP-IDF installation phase (first and second bullet) can be done by downloading the Windows executable from Espressif, I know that... I just posted the manual process to be more generic.

lbernstone, 2020-08-12:
lib-builder will work on windows in msys2.  I would never recommend using Windows to build this stuff (or anything, frankly).  If you do it more than once, it is faster to make a vm of linux and run it in there.  I thought there was a branch in lib-builder for idf4.2, but it looks like that is not the case ATM.

athusrv, 2020-08-12:
I agree that Windows is not the best choice for development, but for those who have Windows 10,  I'd recommend [Microsoft WSL](https://docs.microsoft.com/pt-br/windows/wsl/install-win10) or [WSL2](https://docs.microsoft.com/pt-br/windows/wsl/install-win10#update-to-wsl-2) instead of msys2 (which is very odd in my opinion).

In my case, I'm not lucky, since I'm using Windows 7. I would have to go for msys2, but honestly, I don't want that :) . But I'm glad that are some efforts to make it work for Windows OS too.

closed 2020-08-12 by athusrv