espressif/arduino-esp32 #4249: External Wakeup Interrupt ext0 & ext1 not working together

w4ilun, 2020-08-12:
### Hardware:
Board:							ESP32-PICO-D4 - Custom Board
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.12
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
When using ext0 and ext1 together, only ext0 works to wake from deep sleep, ext1 is ignored. All pins used are RTC GPIOs.

### Sketch:
```cpp

#define BTN_0 GPIO_NUM_33
#define BTN_1 GPIO_SEL_26
#define BTN_2 GPIO_SEL_25
#define BTN_3 GPIO_SEL_32
#define BTN_4 GPIO_SEL_4
#define BTN_PIN_MASK BTN_1|BTN_2|BTN_3|BTN_4

void setup()
{
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0: interruptHandler0(); break;
    case ESP_SLEEP_WAKEUP_EXT1: interruptHandler1(); break;
    default: somethingElse();
  }

  esp_sleep_enable_ext0_wakeup(BTN_0, 0); //BTN 0 is active low
  esp_sleep_enable_ext1_wakeup(BTN_PIN_MASK, ESP_EXT1_WAKEUP_ANY_HIGH); //BTN1-4 is active high
  esp_deep_sleep_start();
}

void loop(){}

void interruptHandler0(){
  //do something like blink LED, this works
}

void interruptHandler1(){
  //do something like blink another LED, this doesn't work unless esp_sleep_enable_ext0_wakeup is commented out
}

void somethingElse(){
  //something else
}

```

atanisoft, 2020-08-12:
your `BTN_PIN_MASK` is incorrectly defined. Try:
```
#define BTN_PIN_MASK (1 << 26) | (1 << 25) | (1 << 32) | (1 << 4)
```
Example: https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/examples/system/deep_sleep/main/deep_sleep_example_main.c#L148-L156

w4ilun, 2020-08-12:
Thanks @atanisoft , gpio.h shows

```
#define GPIO_SEL_26             (BIT(26))  
```

so I thought it would be the same as 1 << 26

I found some additional faults in my hardware which are now resolved.

closed 2020-08-12 by w4ilun

atanisoft, 2020-08-12:
@w4ilun you are right, that does look correct. It is not clear why they do not use that in their examples unfortunately.