espressif/arduino-esp32 #4252: Simple web server keeps crashing

nathan-casabieille, 2020-08-13:
### Hardware:
Board: ESP-WROOM-32
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Arch Linux

Hello,

I am working on a project where each esp32 must host a web server in order to be able to configure them. The problem is that it is absolutely not stable, they keep crashing.

So I reduced my code to the minimal code that allows to create a web server. When I run this code, the esp32 creates the access point but when I try to connect to it, the esp32 crashes.

If I remove the web server part of the code everything works fine, but when I add it it crashes.

Here is the code I am executing:

```
#include <WiFi.h>
#include <WebServer.h>

const char *ssid = "ROOMBOX-1234";
const char *password = "12345678";

IPAddress local_ip(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);

WebServer server(80);

void createAccessPoint()
{
    WiFi.softAP(ssid, password);
    WiFi.softAPConfig(local_ip, gateway, subnet);
    delay(100);
}

void setup() {
    Serial.begin(115200);

    createAccessPoint();

    server.on("/", handle_OnConnect);
  
    server.begin();
}

void loop() {
    server.handleClient();
}

void handle_OnConnect() {
    server.send(200, "text/html", SendHTML()); 
}

String SendHTML() {
    return("<!doctype html><html><head></head><body><p>Hello</p></body></html>");
}
```
And here is the crash log I get when I try to connect to the access point:

```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80114071  A1      : 0x3ffb3d30  
A2      : 0x3ffcd234  A3      : 0x3ffcd468  A4      : 0x3ffccaf8  A5      : 0x3ffcd214  
A6      : 0x0204a8c0  A7      : 0x6504a8c0  A8      : 0x80113f14  A9      : 0x3ffb3cf0  
A10     : 0x3ffcd244  A11     : 0x3ffcd468  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x4011406e:0x3ffb3d70 0x40120da1:0x3ffb3d90 0x40125da9:0x3ffb3dd0 0x4012b046:0x3ffb3df0 0x40114993:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

If anyone has any idea of ​​the cause of this problem, I am interested. Thank You !

lbernstone, 2020-08-13:
Pass pointers, not strings.  Since you didn't decode your exception, I can only guess, but if you put the string directly in the send instead of passing around stack objects, it should not crash.

nathan-casabieille, 2020-08-13:
Indeed that was the problem, thank you for taking the time to answer me, it helps me a lot!

closed 2020-08-13 by nathan-casabieille