espressif/arduino-esp32 #4256: Using the file.write() (with SPIFFS) it won't prompt full.

LusterWong, 2020-08-14:
### Hardware:
Board:							ESP32_DevKitc_V4
Core Installation version:			1.0.4
IDE name:						sloeber
Flash Frequency:					80Mhz
PSRAM enabled:                                     yes
Upload Speed:				       115200
Computer OS:                                        Windows 7

### Description:
Space allocation was 1.9MB APP 190KB SPIFFS.
The program is very simple, after creating a file, keep writing 10 bytes of data until it is full.
I use file.write() and expect that the return value will not equal the number of writes when full, but the result is never full. It keeps going in an endless loop.

### Code:
```
	uint32_t i = 0;
	uint32_t size;

	Serial.printf("Appending to file: %s\r\n", path);

	File file = fs.open(path, FILE_APPEND);
	if (!file)
	{
		Serial.println("- failed to open file for appending");
		return;
	}

	i = 0;
	while(1)
	{
		size = file.print("0123456789");
		if(size == 10)
		{
			i += 10;
		}
		else
		{
			break;
		}
	}
```
Are there any Settings I haven't turned on that are causing this problem?

lbernstone, 2020-08-14:
tl;dr
You must flush or close a file after writing for the change to be committed to disk.

LusterWong, 2020-08-15:
> tl;dr
> You must flush or close a file after writing for the change to be committed to disk.

Need to close and reopen the file after each write?
```
        while(1)
	{
		file = fs.open(path, FILE_APPEND);
		if (!file)
		{
			Serial.println("- failed to open file for appending");
			return;
		}

		size = file.print("0123456789");
		file.close();

		if(size > 0)
		{
			i += 10;
		}
		else
		{
			break;
		}
	}
```
Like the code above?
What is the maximum number of bytes that can be written at a time?
I haven't tested the above code yet, but opening and closing files so often is slow, and does it cause a heap overflow?

lbernstone, 2020-08-15:
file.flush()

LusterWong, 2020-08-15:
> file.flush()

file.flush() Instead of file.close()
```
        uint32_t i = 0;
	uint32_t size;

	File file;

	file = fs.open(path, FILE_APPEND);
	if (!file)
	{
		Serial.println("- failed to open file for appending");
		return;
	}

	i = 0;
	while(1)
	{
		size = file.print("0123456789");
		//file.close();
		file.flush();

		if(size > 0)
		{
			i += 10;
			printf("used size = %d\n", SPIFFS.usedBytes());
		}
		else
		{
			break;
		}
	}

	file.close();
}
```
SPIFFS occupy 190 KB.
An hour went by and it wasn't over yet.

closed 2020-10-28 by stale[bot]