espressif/arduino-esp32 #4261: Serial conflicts with esp_log

ahinsen, 2020-08-16:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).

### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation version:			1.8.12
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    
Partition scheme:					Default                                   
Upload Speed:					921600
Computer OS:                                        Windows 10
### Description:
Hi, 
When I call Serial.println() right after a significant amount of log writing, the message will not go through to the serial monitor. (Significant=4 subsequent esp_log calls as in the example below. With only 3 calls, the problem doesn't occur.)
What is causing this? Is this a bug, or a feature?
If feature, then how do I make sure all my messages get out over the UART? 
(I mean any nicer solution then just applying intuitive delays)

### Sketch:  
```cpp
#define LOG_TAG "tst";
void setup() {
  Serial.begin(112500);   
  while (!Serial);   
  ESP_LOGE( LOG_TAG, "Error:%d", 1 ); 
  ESP_LOGW( LOG_TAG, "Warning:%d", 2 );
  ESP_LOGI( LOG_TAG, "Info:%d", 3 ); 
  ESP_LOGD( LOG_TAG, "Debug:%d", 4 );
  for (int i=0;i<10;i++) Serial.println(i);
}

void loop() {
  ESP_LOGE( LOG_TAG, "Error:%d", 100 ); 
  ESP_LOGW( LOG_TAG, "Warning:%d", 200 );
  ESP_LOGI( LOG_TAG, "Info:%d", 300 ); 
  ESP_LOGD( LOG_TAG, "Debug:%d", 400 );
  for (int i=0;i<10;i++) Serial.println(i);
  delay(5000);
  ESP_LOGE( LOG_TAG, "Error:%d", 100 ); 
  ESP_LOGW( LOG_TAG, "Warning:%d", 200 );
  ESP_LOGI( LOG_TAG, "Info:%d", 300 ); 
  for (int i=0;i<10;i++) Serial.println(i);
  while(1);
}
```

### Output Messages on Arduino IDE's own serial monitor (with timestamp switched on):
```
10:14:21.478 -> [E][log_test.ino:6] setup(): Error:1
10:14:21.478 -> [W][log_test.ino:7] setup(): Warning:2
10:14:21.478 -> [I][log_test.ino:8] setup(): Info:3
10:14:21.478 -> [D][log_test.ino:9] setup(): Debug:4
10:14:21.478 -> 4
10:14:21.478 -> 5
10:14:21.478 -> 6
10:14:21.478 -> 7
10:14:21.478 -> 8
10:14:21.478 -> 9
10:14:21.478 -> [E][log_test.ino:14] loop(): Error:100
10:14:21.478 -> [W][log_test.ino:15] loop(): Warning:200
10:14:21.478 -> [I][log_test.ino:16] loop(): Info:300
10:14:21.478 -> [D][log_test.ino:17] loop(): Debug:400
10:14:21.478 -> 
10:14:21.478 -> 7
10:14:21.478 -> 8
10:14:21.478 -> 9
10:14:26.499 -> [E][log_test.ino:20] loop(): Error:100
10:14:26.499 -> [W][log_test.ino:21] loop(): Warning:200
10:14:26.499 -> [I][log_test.ino:22] loop(): Info:300
10:14:26.499 -> 0
10:14:26.499 -> 1
10:14:26.499 -> 2
10:14:26.499 -> 3
10:14:26.499 -> 4
10:14:26.499 -> 5
10:14:26.499 -> 6
10:14:26.499 -> 7
10:14:26.499 -> 8
10:14:26.499 -> 9
```
Thank you for your help
Andras

lbernstone, 2020-08-17:
logging is not buffered.  serial.flush() will write out the buffer immediately.

ahinsen, 2020-08-17:
Thank you, I understand your reply in its own, but I don't understand how does it answer my questions.
Replacing each `Serial.print(...);` with `Serial.print();Serial.flush();` improves the code, but it still swallows the first print() after the error messages.
Replacing each `Serial.print(...);` with `Serial.flush();Serial.print();` solves the problem.
For me this implies that the buffer is full after the error messages when first print() statement tries to write to it. I don't understand how the buffer can be full if logging is not buffered... or do I get something wrong?

closed 2020-10-31 by stale[bot]

ahinsen, 2024-07-04:
> This is still an issue I have the exact same problem. @ahinsen
> <https://github.com/ahinsen> did you manage to overcome this problem?

Hi, I ended up writing my own logging logic :(