espressif/arduino-esp32 #4268: compilation fails with invalid assembler code: 'lsiu'

neddie766, 2020-08-18:
Arduino 1.8.12
Linux Mint 19.3
esp32 version 1.0.4
board DOIT ESP32 DEVKIT V1

/tmp/ccva6bq6.s: Assembler messages:
/tmp/ccva6bq6.s:58: Error: unknown opcode or format name 'lsiu'

I'm not uploading, just verifying.
'lsiu' seems to be a register name so it would seem something is generating bad assembler code.
I can see the file /tmp/ccva6bq6.s appear for half a second but the ide immediately deletes it so I can't look at it.
In the code below, if I uncomment return 0 and comment out the line immediately above, then it compiles.

`typedef struct CalibrationPoint_ {
	float cRms;
	float amps;
} CalibrationPoint;

#define MAX_CALIBRATIONPOINTS 20
CalibrationPoint calibrationPoints[MAX_CALIBRATIONPOINTS];
uint32_t nCalibrationPoints = 0;

float getCalibrated(float c) {
	if (nCalibrationPoints <= 1) return 0;
	uint32_t iCal = 0;
	while ((iCal < (nCalibrationPoints - 2)) && (c > calibrationPoints[iCal + 1].cRms)) ++iCal;
	CalibrationPoint *const cpsA = calibrationPoints + iCal, *cpsB = cpsA + 1;
	return cpsA->amps + (cpsB->amps - cpsA->amps) * (c - cpsA->cRms) / (cpsB->cRms - cpsA->cRms);
//	return 0;
}

void setup() {
}

void loop() {
}`

Apparently github considers indentation unnecessary.

lbernstone, 2020-08-18:
Three backticks for code (like the link in the [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md)
Perhaps you should set a variable for that value and return that instead of making the compiler guess what the type is.

neddie766, 2020-08-18:
The function has return type float, and all the variables are floats, so there's no guessing.

closed 2020-10-31 by stale[bot]