espressif/arduino-esp32 #4270: ESP32 Wake stub get elapsed time

mightChamp, 2020-08-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:	 16/jun/2020
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
when wake stub starts I want to know how much time ESP32 spend in  deep sleep, how can I achieve this I tried below code in wake stub but always get 0 in return.

### Sketch:  (
```
 static const char RTC_RODATA_ATTR now_str[] = "now=%ld\n";
// Get RTC calibration
  uint32_t period = REG_READ(RTC_SLOW_CLK_CAL_REG);
  // Calculate sleep duration in microseconds
  int64_t sleep_duration = (int64_t)duration_us - (int64_t)DEEP_SLEEP_TIME_OVERHEAD_US;
  if (sleep_duration < 0) {
    sleep_duration = 0;
  }
  // Convert microseconds to RTC clock cycles
  int64_t rtc_count_delta = (sleep_duration << RTC_CLK_CAL_FRACT) / period;
  // Feed watchdog
  REG_WRITE(TIMG_WDTFEED_REG(0), 1);
  // Get current RTC time
  SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);
  while (GET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_VALID) == 0) {
    ets_delay_us(1);
  }
  SET_PERI_REG_MASK(RTC_CNTL_INT_CLR_REG, RTC_CNTL_TIME_VALID_INT_CLR);
  uint64_t now = READ_PERI_REG(RTC_CNTL_TIME0_REG);
  
  now |= ((uint64_t)READ_PERI_REG(RTC_CNTL_TIME1_REG)) << 32;

  ets_printf(now_str,now); // print always o 
```

lbernstone, 2020-08-19:
This forum is for issues with this codebase.  Please ask general questions at https://esp32.com

mightChamp, 2020-08-20:
> This forum is for issues with this codebase. Please ask general questions at https://esp32.com

thanks for kind support

closed 2020-08-20 by mightChamp

lilalukas, 2020-09-21:
> > This forum is for issues with this codebase. Please ask general questions at https://esp32.com
> 
> thanks for kind support

Have you found a solution. I also want to read the time between two interrupts in wake stub and dont have a solution yet...