espressif/arduino-esp32 #4271: ESP32-camera driver update

keyfour, 2020-08-19:
### Hardware:
Board:							ESP32-Cam
Core Installation/update date:			20/May/2020
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     yes
Upload Speed:				        115200
Computer OS:                                         Arch

### Description:

I would like to use [``set_res_raw``](https://github.com/espressif/esp32-camera/blob/master/driver/include/sensor.h#L186) function from last ``esp32-camera`` driver version with Espressif Arduino SDK. Are there any chances to update SDK to use it with Platform.io IDE?

lbernstone, 2020-08-19:
idf-release/v4.0 branch has it
https://github.com/espressif/arduino-esp32/blob/idf-release/v4.0/tools/sdk/include/esp32-camera/sensor.h

keyfour, 2020-08-26:
Thanks @lbernstone , I fixed undefined references using newest toolchain:

```
[env:esp32cam]
board = esp32cam
framework = arduino
platform = espressif32
platform_packages =
	; use a special branch
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
	toolchain-xtensa32 @ 3.80200.200512
```

But anyway I have an issue with ``iram0.text`` section.

```
/home/keyfour/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32cam/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/keyfour/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/keyfour/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 7940 bytes
```

I suppose that happens because of differences in ``esp32.project.ld`` linker scripts:

1. [esp32.project.ld (master)](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/ld/esp32.project.ld#L165l)
2. [esp32.project.ld (idf-release/v4.0)](https://github.com/espressif/arduino-esp32/blob/idf-release/v4.0/tools/sdk/ld/esp32.project.ld#L165)

Any thoughts about this?

lbernstone, 2020-08-26:
No, but I would assume there was some reason @me-no-dev removed the driver from the esp32s2 branch.  The camera driver is a tight fit, and he will likely need to do some rearrangement to get enough iram with IDF 4.

keyfour, 2020-08-28:
I used dirty trick and just replace  ``esp32.project.ld`` by the version from master branch. Now it builds without errors. I will test the resulting firmware with a hardware maybe next week.

Here is my new configs:

```
[env:esp32cam]
board = esp32cam
framework = arduino
platform = espressif32
platform_packages =
	; use a special branch
	framework-arduinoespressif32 @ https://github.com/keyfour/arduino-esp32.git#idf-release/v4.0-iram-issue-fix
	toolchain-xtensa32 @ 3.80200.200512
```

keyfour, 2020-09-01:
I got nonworking firmware naturally, I hide [this comment](https://github.com/espressif/arduino-esp32/issues/4271#issuecomment-682624965) to do not mislead others.

closed 2020-11-15 by stale[bot]