espressif/arduino-esp32 #4287: I have been wanting to enhance sntp.c for a while by adding an 8-bit "reachability" byte.  Each time an NTP packet is successfully received you shift the byte one bit left and "or" in a "1" for the new least-significant bit. Or, if sntp's timer expires without a packet being received you left shift with a 0 as the new LSB. Thus this simple one-byte variable provides the history for the last 8 NTP synchronization requests. (This idea is not mine. The Unix/Linux versions of ntp have been doing this since 1994.)

Clickau, 2020-08-23:
I have been wanting to enhance sntp.c for a while by adding an 8-bit "reachability" byte.  Each time an NTP packet is successfully received you shift the byte one bit left and "or" in a "1" for the new least-significant bit. Or, if sntp's timer expires without a packet being received you left shift with a 0 as the new LSB. Thus this simple one-byte variable provides the history for the last 8 NTP synchronization requests. (This idea is not mine. The Unix/Linux versions of ntp have been doing this since 1994.)

I've added the necessary code and will download it tonight into one of my ESP32s and test it our for a few of days. If all goes well I'll post up the changes here.

_Originally posted by @tferrin in https://github.com/espressif/arduino-esp32/issues/1114#issuecomment-367525304_

lbernstone, 2020-08-23:
The SNTP app is [upstream](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c) in the IDF. It will be easiest to modify there.

tferrin, 2020-08-23:
An 8-bit server "reachability" register was added to the lwIP source code back in March 2018, along with a function to fetch its value: u8_t sntp_getreachability(void). But I don't know if this new version was ever incorporated into the esp32-idf, which then would eventually get included in the arduino-esp32 repository. It seems like enough time has passed for all that to happen but you need to look at source code to know for sure.

Clickau, 2020-08-24:
> The SNTP app is [upstream](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c) in the IDF. It will be easiest to [modify](https://github.com/espressif/esp-idf/pulls) there.

Oops, I didn't even realise I posted this in the Arduino repository. Sorry.

> An 8-bit server "reachability" register was added to the lwIP source code back in March 2018, along with a function to fetch its value: u8_t sntp_getreachability(void). But I don't know if this new [...]

Ok, thanks. I will look into it.

closed 2020-08-24 by Clickau

lbernstone, 2020-08-24:
It'll be in IDF 4.2.  Already available in esp32s2 branch.
https://github.com/espressif/arduino-esp32/blob/esp32s2/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/apps/sntp.h#L62