espressif/arduino-esp32 #4300: Adafruit Feather ESP32 WROOM ISSUE

josetomas, 2020-08-28:
### Hardware 
Adafruit feather ESP32 WROOM - 32 
IDE: Platformio 
Upload Speed: 115200
Computer OS: Windows
Core Installation version: No idea, as it came from fabric
Flash frequency: 40 MHz

### Description
I don't know what happened, the board was working perfectly in the morning, the code was perfect. I was doing some little modifications in order to upload documentation, and the board wasn't working. 
I can compile the codes but they don't work, the problem repeats with every code. 
I tried to erase flash and it didn't work. 

### Sketch 

Same problem with different sketch I had tested and worked before. 

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

```

lbernstone, 2020-08-28:
That is a boot sequence, not an error.  Perhaps you did not call Serial.begin, if you are expecting some output.

josetomas, 2020-08-28:
I called Serial.begin. I not being able to go past this boot sequence. 

This is one the codes I have been uploading for example: 

```
#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "MIT";
//const char* password = "lololo111";

const char* serverName = "http://18.21.85.70/UP_Client";

String httpGETRequest(const char* server);

void setup() {
Serial.begin(115200);
//WiFi.begin(ssid,password);
WiFi.begin(ssid);
while(WiFi.status() != WL_CONNECTED) { 
    delay(500);
    Serial.print(".");
}
Serial.println("");
Serial.print("Connected to WiFi network with IP Address: ");
Serial.println(WiFi.localIP());
}

void loop() {
  Serial.println(httpGETRequest(serverName));
}

String httpGETRequest(const char* server) {
  HTTPClient http;
    
  // Your IP address with path or Domain name with URL path 
  http.begin(server);
  
  // Send HTTP POST request
  int httpResponseCode = http.GET();
  
  String payload = "--"; 
  
  if (httpResponseCode>0) {
    //Serial.println(httpResponseCode);
    payload = http.getString();
  }
  else {
  }
  // Free resources
  http.end();

  return payload;
}
```

lbernstone, 2020-08-28:
Put in a serial.print statement before you init WiFi so you can see that the device is booting.
I'm guessing you do not have adequate power to even get a brownout error.  Plug into a different USB port or use an external power source.

josetomas, 2020-08-28:
I'm sorry, my bad. 

Im using platformio, and I changed the folder location, and IDE lost the track of the other files, so it never loaded the code correctly.

closed 2020-08-28 by josetomas