espressif/arduino-esp32 #4302: ESP32 Bluetooth classic serial to ELM327 dongle no communication

digidax, 2020-08-28:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Joy-IT Node MCU ESP32 dev Board
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                         Windows 10

### Description:
When I use the Android App "serial bluetooth terminal", the ELM327 (OBDII dongle) is listed as BT Classic Device.
I can connect and send AT commands, get also a proper reply.

With ESP32 I have used this example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino

Have tried to use the name "OBDII" for connection init, then the BT Mac address (seen in serial BT terminal) and also enabled the pin in line 24. When I type a command in serial monitor, no reaction.

### Debug Messages:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[I][BluetoothSerial.cpp:510] _init_bt(): device name set
The device started in master mode, make sure remote BT device is on!
[I][BluetoothSerial.cpp:225] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:722] connect(): master : remoteAddress
[I][BluetoothSerial.cpp:290] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT
Connected Succesfully!
[I][BluetoothSerial.cpp:735] connect(): master : remoteAddress
[I][BluetoothSerial.cpp:290] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT
```

digital-spinner, 2020-09-01:
Same here, no packet send? There is no blinks indicating that ELM327 device received anything.

digidax, 2020-09-02:
Have using now [this](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino) example and this is the **not working** result I get in serial monitor:
`mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[I][BluetoothSerial.cpp:510] _init_bt(): device name set
[I][BluetoothSerial.cpp:225] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:702] connect(): master : remoteName
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
Couldn't connect to OBD scanner - Phase 2`

This issue is also monitored [here](https://github.com/PowerBroker2/ELMduino/issues/42)
My ELM dongle is olso working fine with [ELM OBD Terminal](https://play.google.com/store/apps/details?id=com.app.elm327terminal)

rp3tya, 2020-10-31:
I have a similar problem, using SerialToSerialBTM.ino I try to connect to a bluetooth relay (https://www.tinyosshop.com/smartphone-controlled-switch-lazybone-v3-bluetooth). Connection appears successful, but it isn't because behaviour is the same even with a wrong mac address or pin. Data is not exchanged. My conclusion is that currently BluetoothSerial does not work in master mode. Was also discussed in #3916.

closed 2021-01-14 by stale[bot]