espressif/arduino-esp32 #4304: How the BLE or Wifi can be added to pSRAM ?

Sanketa54, 2020-08-28:
Hi,
I'm using Wifi, mqtt  and BLE at the same time .
I'm facing low heap issue.
If pSRAM is added , does BLE stack and Wifi and mqtt libraries can be stored in external psram ?If yes How we can do that

lbernstone, 2020-08-28:
You can use the esp32s2 branch, which has psram mapped into the main memory allocator.
The real solution is to migrate to https://github.com/h2zero/NimBLE-Arduino, which uses half as much memory as bluedroid.

Sanketa54, 2020-08-31:
@lbernstone :
I'm using NimBLE already. But due to wifi and mqtt  i'm loosing some heap.
So, in this branch u mentioned above ....does it adds contagious memory to ESP32 internal memory or just moves the some large allocation to pSRAM.

closed 2020-11-14 by stale[bot]