espressif/arduino-esp32 #4305: RSA public decrypt and private encrypt support?

radimkohout, 2020-08-28:
Does the mbedtls library support RSA decryption with the public key and encrypt with the private key?

lbernstone, 2020-08-28:
mbedtls has hardware RSA accelerators.  You can look in the mbedtls documentation for more information. https://tls.mbed.org/api/
"RSA decryption" is a very wide subject.  Figure out what you really are trying to do and then google for examples.

radimkohout, 2020-08-28:
I have text, that I want to encrypt with RSA private key and decrypt it with RSA public key.

lbernstone, 2020-08-28:
Good luck.  Perhaps you mean something like this. https://github.com/ewpa/LibSSH-ESP32

radimkohout, 2020-10-28:
This is not possible in the library, that you are writing https://github.com/ewpa/LibSSH-ESP32/issues/4

radimkohout, 2021-01-06:
I'm still wondering, if it is possible with mbed API... I need to encrypt using private RSA key and decrypt with public RSA key.

kahotv, 2021-06-11:
me too,i need to encrypt a config file using private key

Elmartino4, 2021-09-03:
it'd be worth looking at [this file](https://github.com/ARMmbed/mbedtls/blob/development/include/mbedtls/rsa.h) from mbedtls sources, it seems to have a similar syntax to [this tutorial on mbedtls aes](https://techtutorialsx.com/2018/04/18/esp32-arduino-encryption-using-aes-128-in-ecb-mode/) for a direction to move in.
I haven't personally looked into comparing the two that much it just seems close after skimming over the two.

VojtechBartoska, 2022-03-30:
Hello @radimkohout, is this still actual?

radimkohout, 2022-03-30:
Yeah, it's still relevant.

closed 2025-01-16 by Parsaabasi