espressif/arduino-esp32 #4308: Turn pins with SDWebserver

mariliahoshino, 2020-08-30:
I like this
https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/SDWebServer/SdRoot

I tried with this
https://github.com/mariliahoshino/Protocol_CAN_interface/blob/master/index%20-%20Copia.htm

but how turn on/off the pins with html in SDcard
I'd like read pins and send data in pins

Method is different than html page is inside the code

<img scr="https://github.com/mariliahoshino/Protocol_CAN_interface/blob/master/Capturar.PNG?raw=true">

lbernstone, 2020-08-30:
This forum is for issues with the code in the repository.  Please ask tutorial questions at https://esp32.com

mariliahoshino, 2020-08-30:
> This forum is for issues with the code in the repository. Please ask tutorial questions at https://esp32.com

The first link is code in the repository and the second is code in my profile with change

closed 2020-11-12 by stale[bot]