espressif/arduino-esp32 #4309: AttachInterrupt FALLING problem

vladigg54, 2020-08-30:
Make your question, not a Statement, inclusive. Include all pertinent information:

Reading code from Wiegand reader/keypad

ESP32 with Hikvison Wiegand Reader/keypad and level shifter BSS138 (in same configuration with ESP8266 no problem)
When press key from keypad or read 125KHz card sometime read correct bits, sometimes missed or added bits.
No error messages, just wrong code read.

I try different library for Wiegand protocol, but with everyone is aam problem.

The code I paste is example from library: https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino

Whe I edit Wiegand.cpp and change interrupt from FALLING to RISING, the problem is gone. Every time I press key from keypad, read this key correct. When I read card, read code from card correct, but when I replace wiegand.cpp with original from library with FALLING, the problem is still there. I try exactly same scenario with another libraries and result is same.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE ,Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Windows 10

#include <Wiegand.h>

WIEGAND wg;

void setup() {
	Serial.begin(115200);  
	
	// default Wiegand Pin 2 and Pin 3 see image on README.md
	// for non UNO board, use wg.begin(pinD0, pinD1) where pinD0 and pinD1 
	// are the pins connected to D0 and D1 of wiegand reader respectively.
	wg.begin(34,35);
}

void loop() {
	if(wg.available())
	{
	Serial.print("Wiegand HEX = ");
		Serial.print(wg.getCode(),HEX);
		Serial.print("DECIMAL = ");
		Serial.print(wg.getCode());
		Serial.print(", Type W");
		Serial.println(wg.getWiegandType());    
	}
}

DougArmstrong, 2020-09-18:
I suggest you read this.  To move forward successfully, you should 1)understand the pull up issues on certain I/O pins  2) understand the WIFI on restrictions and 3) there is a bug using the pins you are using.  This should help: https://github.com/espressif/arduino-esp32/issues/1111#issuecomment-653314383

Once you address the issues discussed here you should have no problem, I am capturing complex wave forms without issue.

vladigg54, 2020-09-19:
I read https://github.com/espressif/arduino-esp32/issues/1111#issuecomment-653314383 , but I can't understand.Is enough to change pin used for wiegand communication != 34~39, or I must add some resistors for D0 and D1 for pull-up.What do you mean about my problem and the second point "2) understand the WIFI on restrictions"?

I have another problem with my project. I have ESP32 with relay controlled by transistor on pin IO27, some LEDS on pin IO32 and IO33, wiegand reader (on problematic 34,35 pins, I will change this pins), SIM800 module connected to pin IO16 and IO17 for 2G communication when WiFi is gone and last RTC 1307 connected to IO21 and IO22. The idea is when I insert code on wiegand keypad and this code is accepted in period (get from RTC), set relay to high and open the door. I have problem with RTC. After 10-15 get time from RTC, RTC stop working and return everytime one same time. When RTC stop, I connect it to another ESP32 only with RTC and upload stock RTCLib (library) example to displqy in 3 sec time now. But RTC is still stop and return same time every time. When I upload example of RTClib wiht set time function RTC start and return everytime correct time. I try with 5 different RTC 1307, but same problem. Maybe problem is again with pin 34, 35 for wiegand, because get time is exactly after read wiegand interrupts in my code? 

Sorry for my English and long long post!
Thank you in advance!

DougArmstrong, 2020-09-19:
Not sure I can provide any feedback on the RTC issues I have not done much with the RTC on ESP32.  The I2C code on ESP32 is pretty solid, if you have not already here is some info on the 1307, it might be helpful: https://www.exploreembedded.com/wiki/Interfacing_DS1307(RTC)_with_PIC16F877A  It sounds like you have ISR timing issues that are causing problems.  

I have found the ADC1 pins do not make good interrupt input pins and I think moving away from using those for ISRs with or without pull ups is a good plan.  You can expect to get spurious interrupts using pins 33&34.  Generally for I/O pin usage this is a good quick reference: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Wish I could offer more feedback.

vladigg54, 2020-09-19:
I read same documentation for pinouts. I will move wiegand to pin 18 and 19, whether it will be okay?

closed 2020-12-03 by stale[bot]